CANopenNode/CANopenNode #392: SRDO support does not build in latest master (v4.0)

ahelten, 2022-04-27:
I need SRDO support (for lifeguard feature) but it appears to have been removed between v2.0 and v4.0 so that it not longer compiles. I get compiler errors about `CO_SDO_t`:

```
CANopenNode/304/CO_SRDO.h:125:5: error: ‘CO_SDO_t’ does not name a type; did you mean ‘CO_TPDO_t’?
  125 |     CO_SDO_t               *SDO;                 /**< From CO_SRDO_init() */
      |     ^~~~~~~~
      |     CO_TPDO_t
```

Are there any plans to resurrect this code? Is it as simple as adding the `CO_SDO_t` type back into the latest code?

Thanks,
Andy

ahelten, 2022-05-03:
It seems this project is dead or dying?

CAN anyone recommend an alternative open source or proprietary CANopen library that supports Linux? I've not found much in my previous searches.

Or better yet, is there a way to use what working functionality in CANopenNode to support life or node guarding? Or is life/node guarding a sufficiently outdated feature of CANopen that no modern library supports it?

martinwag, 2022-05-03:
> It seems this project is dead or dying?

It's like with all other open source projects where there is not a big company behind. If someone has time, you get help, otherwise not. CANopen (not FD) also has not changed that much for the last 10 years :-)

> CAN anyone recommend an alternative open source or proprietary CANopen library that supports Linux? I've not found much in my previous searches.

Open source: No. 
Proprietary: Just look at their docs. Most of them do.

>  Or is life/node guarding a sufficiently outdated feature

Outdated for a very long time. Check your requirements, if you don't need it, drop it.

ahelten, 2022-05-04:
Thanks for the reply @martinwag , unfortunately it's not about requirements -- the motor controller I'm using only supports life/node guarding. It doesn't support the RPDO event timer nor the heartbeat consumer. I've requested support for the RPDO event timer and they've responded they will _look into_ adding the heartbeat consumer and/or RPDO event timer. Hopefully they follow through but your response adds more weight to my viewpoint.

On the subject of open source projects, I understand the problem completely! I would be willing to implement this support but, like most small companies, I don't really have time and, the fact that life/node guarding is mostly obsolete, makes it hard to find time.

Thanks again!
Andy

CaCtus491, 2022-05-04:
Not that it helps you Andy, but I have the need for SRDO support. It is on
my list to look at adding back in, but quite far down.

Cheers,
Kent

CANopenNode, 2023-03-28:
SRDO is still missing in the new version. Waiting someone willing to cooperate and test.... Closing issue for now.

closed 2023-03-28 by CANopenNode

ali-ghadyali-fvt, 2023-10-10:
> SRDO is still missing in the new version. Waiting someone willing to cooperate and test.... Closing issue for now.

I need this functionality. I can help implement and test.