CANopenNode/CANopenNode #393: Porting CanOpenNode to Teensy 4.x

N5-nfive, 2022-05-03:
Hello,

first I am not sure, if this is the right place for my question. If it is not, I would be happy if you tell me the adequate place. 

For our project at university we are trying to run CanOpenNode on Teensy 4.1. 
We found that there is a version for Teensy3 Board only at `deviceSupport.md`. 

We code and fixed our way until a compiling example based on [Teensy3-code](https://github.com/c0d3runn3r/CANopenNode/tree/add-k20-driver/stack/ARM_Kinetis_K20_teensy) and this [FlexCan](https://github.com/tonton81/FlexCAN_T4) library, we modified a little. 
Right now our teensy4.1 gets lost in a loop at `void CO_CANsetConfigurationMode(int32_t CANbaseAddress)`. 

We assume, the cause for a lot of time we wasted on this and other problems before is our lack of knowledge of what the functions declared in `driver.h` actually have to do.  

Therefore, we would like to know, if there is:
- A more detailed documentation than inline comments about what is required to happen in `driver.c`,
- or: Someone who has time and interest to review our `dirver.cpp` for wrongly configured registers etc.,
- or: Any other help :)

I know this sounds quite general, so feel free to reach out for additional information. If you want to have a look at the code, we can add you to our repository.

Many thanks.

Background: On the team are 4 students with 1-3 years experience in writing C/C++-Code and basic software engineering, to give you an impression what to expect. We have used CanOpen before, but have never dug into internals. We use PlatoformIO and Teensydiuno as a framework and run FreeRTOS on the chip.

ned-pcs, 2022-05-12:
See Mark Butcher's µTasker (https://www.uTasker.com); he has a port of his uTasker operating system that includes a CanOpenNode port for the i.MXRT series MCUs.

closed 2023-03-28 by CANopenNode

owenchen0408, 2023-12-29:
Hi, I am also creating a project using CAN and teensy, do you mind sharing the code?

bikeNomad, 2023-12-29:
See https://www.utasker.com/iMX.html