CANopenNode/CANopenNode #395: CO_driver.c porting clarify

xenpac, 2022-05-24:
i am porting the canopen stack to a device.
i understand that the interface shall be implemented in co_driver.c.

looking at:

_CO_ReturnError_t CO_CANsend(CO_CANmodule_t *CANmodule, CO_CANtx_t *buffer){
    CO_ReturnError_t err = CO_ERROR_NO;

    /* Verify overflow */
    if(buffer->bufferFull){
        if(!CANmodule->firstCANtxMessage){
            /* don't set error, if bootup message is still on buffers */
            CANmodule->CANerrorStatus |= CO_CAN_ERRTX_OVERFLOW;
        }
        err = CO_ERROR_TX_OVERFLOW;
    }

    CO_LOCK_CAN_SEND(CANmodule);
    /* if CAN TX buffer is free, copy message to it */
    if(1 && CANmodule->CANtxCount == **0){**
        CANmodule->bufferInhibitFlag = buffer->syncFlag;
        /* copy message and txRequest */
    }
    /* if no buffer is free, message will be sent by interrupt */
    else{
        buffer->bufferFull = true;
        CANmodule->CANtxCount++;
    }
    CO_UNLOCK_CAN_SEND(CANmodule);

    return err;
}_

This seems to be the hardware CAN-send routine.
The buffers used here is that canopen internal or are these CAN-hardware buffers?
What is the meaning of "if no buffer is free, message will be sent by interrupt" ?

i have my own CAN IO-buffers and just need the canopen interface for input and output of CANframes.

Maybe someone can shed some light on this.

Thank you.

martinwag, 2022-05-24:
Take a look at the linux port for a hardware-less implementation.

xenpac, 2022-05-26:
Thank you,

i am new to canopen so may ask some stupid questions ;)

from the above, i understand that canopen maintains a list of valid buffers with Ids.
Only these are accepted for IO.

From the application point of view, there are no api functions to request IO to/from another device.
Its all OD driven.
IO takes place when the data of an OD entry changes and then related OD entrys are exchanged.
So just updating the data in OD entry will trigger transmission.
(wonder how it detects that)

In my application my master-device needs to query or set data in slaves using SDO transfers with fixed 
record size.

The records are defined in slaves OD.
Do i also have to have a corresponding entry in my master device? (with slave-id?)

could you give me an example on how to send or receive such data from my application (master device)?

That would be very helpfull.

martinwag, 2022-05-27:
CANopen is a protocol on top of CAN, so you don't send/receive plain can messages. Get yourself familiar with the concept of PDO and SDO (either read stuff on the web or get the book https://www.canopenbook.com/ ). This all has nothing to do with canopenNode (the stack), but is common to canOpen.

If you understand how SDO is supposed to work, you can read/write from other devices as explained in https://github.com/CANopenNode/CANopenNode/blob/master/301/CO_SDOclient.h

xenpac, 2022-05-28:
yap, thats what i was looking for ;)

xenpac, 2022-06-05:
i have another question:
in my CO_driver.c, i put debug printf in the Rx and Tx buffer init routines.
i am getting this listing of buffer inits:
++++
CANopenNode Starting up or reinit communication
can configured
RXinit: index:16 id:0x07e5 mask:0x07ff object:0x20003f90  lss slave
TXinit: index:8 id:0x07e4 rtr:0 Bytes:8 sync:§d (should be %d)
TXinit: index:1 id:0x0082 rtr:0 Bytes:8 sync:§d
RXinit: index:0 id:0x0000 mask:0x07ff object:0x200030d8  nmt init nmtreceive
TXinit: index:7 id:0x0702 rtr:0 Bytes:1 sync:§d
RXinit: index:8 id:0x0000 mask:0x07ff object:0x20003140  HBconsumerinit <CO_HBcons_receive>  no ID
RXinit: index:9 id:0x0000 mask:0x07ff object:0x20003150  CO_HBcons_receive
RXinit: index:10 id:0x0000 mask:0x07ff object:0x20003160 CO_HBcons_receive
RXinit: index:11 id:0x0000 mask:0x07ff object:0x20003170 CO_HBcons_receive
RXinit: index:12 id:0x0000 mask:0x07ff object:0x20003180 CO_HBcons_receive
RXinit: index:13 id:0x0000 mask:0x07ff object:0x20003190 CO_HBcons_receive
RXinit: index:14 id:0x0000 mask:0x07ff object:0x200031a0 CO_HBcons_receive
RXinit: index:15 id:0x0000 mask:0x07ff object:0x200031b0 CO_HBcons_receive
RXinit: index:7 id:0x0602 mask:0x07ff object:0x200032b0  SDO server init
TXinit: index:6 id:0x0582 rtr:0 Bytes:8 sync:§d
RXinit: index:1 id:0x0080 mask:0x07ff object:0x20003398  sync init
TXinit: index:0 id:0x0080 rtr:0 Bytes:0 sync:§d
RXinit: index:3 id:0x0000 mask:0x07ff object:0x20003400 rpdo init
RXinit: index:4 id:0x0000 mask:0x07ff object:0x20003574 rpdo init
RXinit: index:5 id:0x0000 mask:0x07ff object:0x200036e8 rpdo init
RXinit: index:6 id:0x0000 mask:0x07ff object:0x2000385c rpdo init
TXinit: index:2 id:0x0000 rtr:0 Bytes:0 sync:§d
TXinit: index:3 id:0x0000 rtr:0 Bytes:0 sync:§d
TXinit: index:4 id:0x0000 rtr:0 Bytes:0 sync:§d
TXinit: index:5 id:0x0000 rtr:0 Bytes:0 sync:§d
My CAN NodeID is:2  (Slave device)
++++
These entrys with ID = 0x000 are irritating me. (except NMT init)
Is that valid? or did i configure something wrong?

CANopenNode, 2022-06-06:
It seems correct. HBcons or PDO may not be active after startup. Depends on configuration in object dictionary.

closed 2023-03-17 by CANopenNode