CANopenNode/CANopenNode #396: CANopenLinux & CANopenSTM32: SDO protocol timed out (ERROR:0x05040000).

G-Sinatra, 2022-06-01:
I'm a student new to Embedded System programming and I'm trying to develop a "Master-Slave" CANopen protocol for a University project. 

The approach consist on:
 - Master device: Linux-based Mircroprocessor with  'canopend+cocomm' interface (STM32MP157C-DK2 with a customized shield board mounting an 'MAX3051ESA+' transceiver). Both executable files have been compiled from 'CANopenLinux' and executed following tutorials without modification.  
 - Slave device: STM32H7-based Evaluation board with 'CANopenSTM32' stack protocol integrated. (This device is identified as 18th node in the network, ID: 0x12)

The communication is set with a nominal communication of 250 kbit/s bitrate and 1000 kbit/s databitrate.
In a preliminary phase the overall system worked by showing correctly the 'candump' frames on Linux terminal and also commands like "1 read 0x1017 0 u16" and "1 write 0x1017 0 u16 1000" are correctly executed. 
A first "obstacle" has been the same commands referred to the "slave" device, since both "18 read 0x1017 0 u16" and "18 write 0x1017 0 u16 1000" commands executed on cocomm (or also on canopend with argument '-c stdio') returns: 

`[1] ERROR:0x05040000 #SDO protocol timed out.`

From a 'candump' point of view both command works, the Slave's hearbeat changes, but from the 'Master (canopend/cocomm)' point of view I cannot visualize a correct response, even enlarging the SDO timeouts on 'CO_main_basic.c'. 
Do I have to manually insert functions inside the code to enable the possibility to "read" these "SDO messages"?

CANopenNode, 2022-06-02:
I tested CANopenLinux only on PC or Raspberry PI running Debian (Ubuntu). There may be some issues between CAN interface and canopend.

With "1 read 0x1017 0 u16", CAN interface is not used.

martinwag, 2022-06-02:
- check your termination (2x120 Ohm)
- Disable the CAN FD stuff and try again with plain CAN (on both devices). You can use 1000kBit if that is really necessary.

G-Sinatra, 2022-06-07:
> * Disable the CAN FD stuff and try again with plain CAN (on both devices). You can use 1000kBit if that is really necessary.

Thank you, disabling the FD configuration on both peripheral did the trick.

closed 2022-06-07 by G-Sinatra