CANopenNode/CANopenNode #400: OD, understand DOMAIN data type

xenpac, 2022-06-09:
I work my way into CANopen ;)

I read the CIA 301, but still dont understand what DOMAIN data type actually is.

I created an OD object of type  VAR and Domain.
In OD.c , data size is 0.

- its an overlay of an existing OD-entry, so be able to read/write the whole array or record in one go?

Also i heard that DOMAIN type SDO transfers  are able to transfer "unlimited" amount of bytes.
hmm
That would mean data is actually transfered to the application.  like file transfer.

im still digging in the dark ..

martinwag, 2022-06-10:
Domain = Stream Transfer / Block Transfer, depending on how you use it.

You have to implement a callback that takes the data. Size limit should be 4GB (because of 32 bit address).

xenpac, 2022-06-16:
ok, SDO transfers usually read/write from/to variables (objects) contained in the OD.

For an SDO transfer you have to specify a target OD-entry.

By "Stream Transfer" you mean that the sender app is sending to the receiver app without the need to have a variable in OD.

So the receiver app must register a callback (CO_SDOserver_initCallbackPre()) to catch the data when it arrives. ?

A typical OD entry for a DOMAIN looks like that:

.o_2011_my_Domain = {
        .dataOrig = NULL,   // pointer to the data area
        .attribute = ODA_SDO_RW,
        .dataLength = 0   //length of data area
    },

Whats the use of such an entry ?

martinwag, 2022-06-17:
You need a OD entry and all the configuration stuff, but it is not used to store data. Look for domain data type in closed questions, this has been asked a few times.

xenpac, 2022-06-17:
Ok, i see, the Domain entry is just a dummy. to have an object address.

The function CO_SDOserver_initCallbackPre has the following parameters:
void CO_SDOserver_initCallbackPre(CO_SDOserver_t *SDO,
                                  void *object,  // this should be my_Domain entry (dummy), where the callback will trigger
                                  void (*pFunctSignalPre)(void *object))  //this shall be my callback function to catch the data to app

So when the master(client) starts an SDO transfer using the my_Domain object, the slave(server) will call my callback function, instead of trying to access some OD entry.

If you have not registered a callback for my_Domain, the server will return an error because the datapointer is NULL.

CANopenNode, 2022-06-20:
CO_SDOserver_initCallbackPre() may be used for wakeup in some operating systems.

Please see https://github.com/CANopenNode/CANopenDemo/blob/master/demo/domainDemo.c for example.

Actually, in the core (new object dictionary principle), each Object dictionary entry is accessed by the two functions OD_readxxx() and OD_writexxx(). By default functions just read/write data defined by in Object Dictionary (OD.c file). But application may register own two functions, which may act like a stream. See https://canopennode.github.io/CANopenNode/md_doc_objectDictionary.html

With segmented or block transfer it is possible to transfer 4GB of data, if size is indicated. If size is not indicated, there is actually no limit, but I don't think, this may be practical. There is no need to have large internal buffer.

xenpac, 2022-06-27:
something like this:
------------
typedef struct {
    uint16_t index;
    uint8_t subEntriesCount;
    uint8_t odObjectType;
    CO_PROGMEM void *odObject;
    OD_extension_t *extension; // see below. if this is NULL, then standard OD io is done.
} OD_entry_t; //= structure of each OD entry

typedef struct {
    void *object;  // this is of type: OD_stream_t and describes the object on which below function can operate.
	               // in user io, this may not be used.
     ODR_t (*read)(OD_stream_t *stream, void *buf, OD_size_t count, OD_size_t *countRead); //user read function
    ODR_t (*write)(OD_stream_t *stream, const void *buf,OD_size_t count, OD_size_t *countWritten);//user write function

} OD_extension_t;

/* this inline function will register a custom IO callback/function for a specific OD entry.
   It will work on any OD object, not only on Domains.
   
 entry:
 - pointer to a specific OD entry on OD.c, where this extension is placed.
 - pointer to an OD_extension_t struct containing function pointers for custom read and/or write function.
   if read and/or write functionpointer is NULL, then this means that the operation is disabled.
   The **object** basicly specifies the datalength and current dataoffset of the transmission.(OD_stream_t)
   The dataoffset is updated during transmission and can be used to identify current state.
   
exit:
- return code: ODR_OK = 0 else failure.   
*/
static inline ODR_t OD_extension_init(OD_entry_t *entry,
                                      OD_extension_t *extension)
{
    if (entry == NULL) return ODR_IDX_NOT_EXIST;
    entry->extension = extension;
    return ODR_OK;
}
--------------------

CANopenNode, 2023-03-28:
I hope, this is solved. If necessary, re-open the issue.

closed 2023-03-28 by CANopenNode