CANopenNode/CANopenNode #402: Emergency message packing is incorrect?

andrew-elder, 2022-06-10:
If I read https://www.can-cia.org/can-knowledge/canopen/special-function-protocols/ it states

```
The data content is defined as a 1 byte Error register (Index 1001h of the local object dictionary), a 16-bit Emergency error code, and up to 5 byte of manufacturer-specific error information. 
```

But the code here https://github.com/CANopenNode/CANopenNode/blob/master/301/CO_Emergency.c#L746 doesn't support access to 5 bytes of manufacturer-specific error information. On the wire I see

```
byte[0] LSB of Error code 
byte[1] MSB of Error code 
byte[2] 0
byte[3] Error register value
...
```

I would have expected byte[2] to be the Error register value. Bytes on the wire were captured using PCAN and the code is running on an ESP32. SDOs reads and writes are working as expected.

martinwag, 2022-06-13:
should be infocode variable https://github.com/CANopenNode/CANopenNode/blob/master/301/CO_Emergency.h#L611

andrew-elder, 2022-06-13:
@martinwag - thanks for the prompt response. If I read https://www.can-cia.org/can-knowledge/canopen/special-function-protocols/ it states ` up to 5 byte of manufacturer-specific error information`. For a project I'm working on, I need to be able to specify all 5 bytes of the manufacturer-specific error information, not just the 4 bytes represented by the `infocode` variable. What are my options?

martinwag, 2022-06-13:
If I remember correctly one byte is used by the stack itself. If you need to use this byte you have to modify the stack source.

andrew-elder, 2022-06-13:
This python implementation shows what I would expect https://github.com/christiansandberg/canopen/blob/master/canopen/emcy.py#L8

I'll create a fork and make some changes for review.

martinwag, 2022-06-13:
I've had a look, one byte is used by the stack. You can change that in your fork, but changing it here will break compatibility.

andrew-elder, 2022-06-13:
Yes, I see that now. The message format is described here
```
 *   Byte | Description
 *   -----|-----------------------------------------------------------
 *   0..1 | @ref CO_EM_errorCode_t
 *   2    | @ref CO_errorRegister_t
 *   3    | Index of error condition (see @ref CO_EM_errorStatusBits_t).
 *   4..7 | Additional informative argument to CO_errorReport() function.
```

andrew-elder, 2022-06-13:
Question answered - closing. Thanks @martinwag

closed 2022-06-13 by andrew-elder