CANopenNode/CANopenNode #404: Master node sends continually a message

warcomeb, 2022-07-14:
Hello,
I am trying to use CANopenNode into a STM32L476 device by using [libohiboard](https://github.com/ohilab/libohiboard) as HAL library. In the network, I have: (i) my board that operates as a master and (ii) a commercial node. At startup, the node sends HB message and SYNC message. When my board use "CO_NMT_sendCommand(CO->NMT,CO_NMT_ENTER_OPERATIONAL, 0x0A);" the master starts to send continually the same message without stopping!
Where is the error?

Thanks
Marco

warcomeb, 2022-07-14:
With logic analyzer I see this:
[image]
Where Channel 0 is the TX pins of the microcontroller, and Channel 1 is the RX pin.

GiulioDallaVecchia, 2022-07-14:
It seems that there is no ACK from commercial node

warcomeb, 2022-07-14:
Ok... but is wrong my packet?

warcomeb, 2022-07-14:
@GiulioDallaVecchia I can't understand why the message returns into RX pin immediately! I checked the microcontroller configuration and the loopback mode is OFF.
[image]