CANopenNode/CANopenNode #409: Synchronization Tests are failed in CTT tool

EIC-VihangJani, 2022-07-26:
I use this code and validating on CANopen conformance test tool when I fail the multiple test cases.
I need your guidance for resolving fail test cases and attach the log file for your reference.
[test_1519_26-07-2022_0001.log](https://github.com/CANopenNode/CANopenNode/files/9188830/test_1519_26-07-2022_0001.log)

martinwag, 2022-07-26:
The canopen test tool has various bugs. Please be aware that my notes are ~5 Years old and the CTT as well as CANopennode have evolved since :-)
- set heartbeat to "0" before testing (disable)
- I've noted that test SYNC 02 has a bug in counting
- with our implementation test SYNC 03 fails, but I didn't investigate this further.

EIC-VihangJani, 2022-07-26:
Where can I set heartbeat timer in code side or in EDS file ?

martinwag, 2022-07-26:
Register 0x1017

ankit-eic, 2022-07-27:
> Register 0x1017

The value of the register 0x1017 is already zero.

Blazej1994, 2022-09-01:
> The canopen test tool has various bugs. Please be aware that my notes are ~5 Years old and the CTT as well as CANopennode have evolved since :-)
> 
> * set heartbeat timer to "0" before testing (disable)
> * I've noted that test SYNC 02 has a bug in counting
> * with our implementation test SYNC 03 fails, but I didn't investigate this further.

Now I am using CTT version 3.0.3.10 - newst one, and SYNC03 also fails for me, but only if before SYNC02 is executed.
If you execute only test SYNC03 after device reset then it succes.
I investigated and I think that I found the reason. In SYNC02 CTT set object 0x1006 Communication cycle period to 151000us = 151 ms (in my case at least), and in this test our controller is SYNC producer. After finishing this test CTT is disabling SYNC producing in our controller, but is not resetin 0x1006 to 0. Because of that our controller, which earlier was using 0x1006 to produce SYNC frame now is using it to check interval of received SYNC frame, and if it exceeds time set in 0x1006 it is reporting error and changing state to PreOperational.

This is happening because finishing SYNC02, CTT is setting controller for SYNC03, and before CTT sends first SYNC frame, time from last frame SYNC sent by our controller (from SYNC02 )is longer that 0x1006 - 150ms. Then our controller reports error, changes state to PreOperational, and doesn't send TPDO on received SYNC frame, because it should send TPDO only in state Operational. 

I think it is not fully proper that controller is checking time interval between last SYNC frame which it produced and first SYNC frame which it should receive - this is checked in function CO_SYNC_process() in CO_SYNC.c.

This checking also implies additional issu. If your controller is SYNC consumer, 0x1006 is equal to 0, and you send SYNC frame to it, and then some time after it you set 0x1006 to some value - time which is longer thant time between sending SYNC frame and now, controller immidiately reports the same error.

I think that there are two ways to fix it.
First: disable checking time interval between received SYNC frames - this checking is not mandatory according to CiA 301, chapter 7.2.5 Synchronization object (SYNC).
Second:
This time interval is checked only if variable SYNC->timeoutError is equal to 1. It is set to 1 every time controller sends or receivs SYNC frame. You can change it to for example 0 when controller changes from SYNC producer to consumer (optionally from consumer to producer as well) and when value in 0x1006 changes.

CANopenNode, 2023-03-28:
Here is my CANopen conformance test tool report:
https://github.com/CANopenNode/CANopenDemo/blob/master/test/README.md#canopen-conformance-test-tool

There are some issues, but not with SYNC.

closed 2023-04-11 by CANopenNode