CANopenNode/CANopenNode #410: Communication error receive

EIC-VihangJani, 2022-07-26:
I receive communication error every time from stack 
Could you give any guidance for it ? and I attach log file for your refe
[Uploading Test_1540_26-07-2022_0002.log…]()
rence

26.07.22 : 15:47:14.042 : Test CAN    : CAN1  606 Tx SD  8  40 01 10 00 00 00 00 00
26.07.22 : 15:47:14.106 : Test CAN    : CAN1  586 Rx SD  8  4F 01 10 00 10 00 00 00
26.07.22 : 15:47:14.106 : Test Error  : OD 02: Accessing object 1001h:00h - Response value: 0x10, EDS default value: 0x00 failed.
26.07.22 : 15:47:14.106 : Test Notice : OD 02: Accessing object 1001h:00h - Response value: 0x10, EDS default value: 0x00 failed.

Bartimaeus-, 2022-07-26:
Your log file didn't attach properly (it's just a link to this issue).

Index 0x1001 is the error register. It looks like bit 7 is set, indicating a manufacturer specific error

[image]

Interestingly CiA 301 specifies that bit 0 (generic error) should also be set when any of the other bits are set (my guess is that behavior isn't implemented by CANopenNode)
[image]

you could try reading from index 0x1003 and see if there's additional info in the error fields:
[image]

EIC-VihangJani, 2022-07-27:
[TEST_1730_26-07-2022_0001.log](https://github.com/CANopenNode/CANopenNode/files/9194985/TEST_1730_26-07-2022_0001.log)

I attached the least file

closed 2023-03-17 by CANopenNode