CANopenNode/CANopenNode #412: How to use CO_MULTIPLE_OD?

Spider84, 2022-08-16:
Right I understand that by defining CO_MULTIPLE_OD I can make multiple devices inside one physically device?
In Example I can see minimal use of CO_MULTIPLE_OD.
How to make two different ODs in one device?

closed 2023-03-17 by CANopenNode

Spider84, 2023-03-18:
Can you make simple example how init two logocal devices inside one physical?

CANopenNode, 2023-03-18:
One CANopen device is already complex, it is not possible to make simple example for other combinations. In general, library is object oriented and you can generate two CANopen devices inside main(). CO_MULTIPLE_OD gives you some helper macros, etc. First you have to understand the details about library and CANopen. It is all documented there.