CANopenNode/CANopenNode #413: How to create dynamic TPDO and RPDO in the code?

ankit-eic, 2022-08-25:
Hi,

I would like to know, how could I create TPDO and RPDO dynamically into the code using API. Is this functionality available in the code?
If it is implemented how I can use it? Any hint or feedback would be nice for understanding code.

Thanks,
Ankit

ankit-eic, 2022-08-29:
Hi,

Could I get any update on the above question? Thanks in advance.

Regards,
Ankit

CANopenNode, 2022-08-29:
PDOs are configured dynamically with correct sequence of writing into object dictionary. See also https://www.can-cia.org/can-knowledge/canopen/pdo-protocol/

ankit-eic, 2022-08-29:
> 

> PDOs are configured dynamically with correct sequence of writing into object dictionary. See also https://www.can-cia.org/can-knowledge/canopen/pdo-protocol/

Thanks for the reply. Is it possible to add new PDOs while the application is running?

CANopenNode, 2022-08-29:
With some advanced programming, maybe yes.

closed 2023-03-17 by CANopenNode