CANopenNode/CANopenNode #418: Porting to RP2040

07spider70, 2022-11-17:
Hi, 
I would like ask, is there a possibility to use this CANOpenStack on RP2040 MCU with external CAN transceiver?
I try understand how port to other device, e.g. PIC work, but I found that most supported devices are with CAN subsystem. 
I want use this stack and transport data over UART to CAN transceiver. I also need FreeRTOS, and I saw that there is some port to it, but is old and I don't understand how send data to GPIO. I need just some example or some initial push to right direction :) 
Thanks a lot.

martinwag, 2022-11-18:
- You have to implement the functions in 301/CO_driver.h . This is not specific to any CAN transceiver. Take a look at the linux example, this works with a serial I/O (socket) and doesn't have any direct access to CAN hardware.
- I've written the FreeRTOS Port. The working principle should not have changed, it just needs updating to the current stack version.
- What do you need GPIO for?

07spider70, 2022-11-19:
ok, I will check the linux example and I will also try to update your FreeRTOS Port to the current stack version.
GPIO are solved already :) I am starting to understand how this stack works. Thanks a lot. 👍

closed 2022-11-19 by 07spider70