CANopenNode/CANopenNode #419: Lining error for Pic33 project

robot2012, 2022-11-23:
Hi,

I downloaded the project and loaded it into MPLAB X IDE V6.05 with xc10 compiler.

I got the following error on my first build. I did a grep for "OD_CANNodeID" in the project and file system and nothing showed up besides where is used.

It looks like I am missing some files. 

Here are the compiler outputs:
```
nbproject/Makefile-default.mk:298: recipe for target 'build/default/production/main_dsPIC33F.o' failed
make[2]: Leaving directory 'E:/Projects/cz/Microchip/X16/CanOpen/dsPIC33FJ256GP710/CANopenPIC/example_dsPIC33_ex16_IO.X'
nbproject/Makefile-default.mk:85: recipe for target '.build-conf' failed
make[1]: Leaving directory 'E:/Projects/cz/Microchip/X16/CanOpen/dsPIC33FJ256GP710/CANopenPIC/example_dsPIC33_ex16_IO.X'
nbproject/Makefile-impl.mk:39: recipe for target '.build-impl' failed
main_dsPIC33F.c: In function 'LSScfgStoreCallback':
main_dsPIC33F.c:94:5: error: 'OD_CANNodeID' undeclared (first use in this function)
main_dsPIC33F.c:94:5: note: each undeclared identifier is reported only once for each function it appears in
main_dsPIC33F.c:95:5: error: 'OD_CANBitRate' undeclared (first use in this function)
main_dsPIC33F.c: In function 'main':
main_dsPIC33F.c:137:8: error: 'CO_OD_RAM' undeclared (first use in this function)
main_dsPIC33F.c:138:8: error: 'CO_OD_EEPROM' undeclared (first use in this function)
main_dsPIC33F.c:139:8: error: 'CO_OD_ROM' undeclared (first use in this function)
main_dsPIC33F.c:143:5: error: too few arguments to function 'CO_new'
../CANopenNode/CANopen.h:419:7: note: declared here
main_dsPIC33F.c:154:21: error: 'OD_CANNodeID' undeclared (first use in this function)
main_dsPIC33F.c:156:22: error: 'OD_CANBitRate' undeclared (first use in this function)
main_dsPIC33F.c:160:5: error: 'OD_powerOnCounter' undeclared (first use in this function)
main_dsPIC33F.c:177:9: error: 'OD_writeOutput8Bit' undeclared (first use in this function)
main_dsPIC33F.c:183:9: warning: passing argument 2 of 'CO_CANinit' makes pointer from integer without a cast
../CANopenNode/CANopen.h:451:18: note: expected 'void *' but argument is of type 'uint16_t'
main_dsPIC33F.c:183:9: error: too few arguments to function 'CO_CANinit'
../CANopenNode/CANopen.h:451:18: note: declared here
main_dsPIC33F.c:187:9: warning: passing argument 1 of 'CO_LSSinit' from incompatible pointer type
../CANopenNode/CANopen.h:469:18: note: expected 'struct CO_t *' but argument is of type 'uint8_t *'
main_dsPIC33F.c:187:9: warning: passing argument 2 of 'CO_LSSinit' from incompatible pointer type
../CANopenNode/CANopen.h:469:18: note: expected 'union CO_LSS_address_t *' but argument is of type 'uint16_t *'
main_dsPIC33F.c:187:9: error: too few arguments to function 'CO_LSSinit'
../CANopenNode/CANopen.h:469:18: note: declared here
main_dsPIC33F.c:192:9: warning: passing argument 1 of 'CO_CANopenInit' makes pointer from integer without a cast
../CANopenNode/CANopen.h:507:18: note: expected 'struct CO_t *' but argument is of type 'uint8_t'
main_dsPIC33F.c:192:9: error: too few arguments to function 'CO_CANopenInit'
../CANopenNode/CANopen.h:507:18: note: declared here
main_dsPIC33F.c:201:46: error: 'CO' undeclared (first use in this function)
main_dsPIC33F.c:249:13: error: 'OD_performance' undeclared (first use in this function)
main_dsPIC33F.c:249:28: error: 'ODA_performance_mainCycleTime' undeclared (first use in this function)
main_dsPIC33F.c:250:35: error: 'ODA_performance_mainCycleMaxTime' undeclared (first use in this function)
main_dsPIC33F.c:255:13: warning: passing argument 3 of 'CO_process' makes integer from pointer without a cast
../CANopenNode/CANopen.h:567:20: note: expected 'uint32_t' but argument is of type 'void *'
main_dsPIC33F.c:255:13: error: too few arguments to function 'CO_process'
../CANopenNode/CANopen.h:567:20: note: declared here
main_dsPIC33F.c: In function '_T2Interrupt':
main_dsPIC33F.c:289:8: error: 'CO' undeclared (first use in this function)
main_dsPIC33F.c:296:9: error: too few arguments to function 'CO_process_RPDO'
../CANopenNode/CANopen.h:609:6: note: declared here
main_dsPIC33F.c:300:24: error: 'OD_writeOutput8Bit' undeclared (first use in this function)
main_dsPIC33F.c:312:9: error: 'OD_readInput8Bit' undeclared (first use in this function)
main_dsPIC33F.c:345:5: error: 'OD_performance' undeclared (first use in this function)
main_dsPIC33F.c:345:20: error: 'ODA_performance_timerCycleTime' undeclared (first use in this function)
main_dsPIC33F.c:346:27: error: 'ODA_performance_timerCycleMaxTime' undeclared (first use in this function)
main_dsPIC33F.c: In function '_C1Interrupt':
main_dsPIC33F.c:356:21: error: 'CO' undeclared (first use in this function)
make[2]: *** [build/default/production/main_dsPIC33F.o] Error 255
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [.build-conf] Error 2
make: *** [.build-impl] Error 2

BUILD FAILED (exit value 2, total time: 3s)

```

File system search results:
```
 E:\Projects\cz\Microchip\X16\CanOpen\dsPIC33FJ256GP710\CANopenPIC\example_dsPIC30F.X\main_dsPIC30F.c (2 hits)
	Line 87:     OD_CANNodeID = id;
	Line 128:     pendingNodeId = OD_CANNodeID;
  E:\Projects\cz\Microchip\X16\CanOpen\dsPIC33FJ256GP710\CANopenPIC\example_dsPIC33_ex16_IO.X\main_dsPIC33F.c (2 hits)
	Line 94:     OD_CANNodeID = id;
	Line 154:     pendingNodeId = OD_CANNodeID;
```
Thank you!

martinwag, 2022-11-24:
Those variables are not part of the stack, they need to be available in your OD (Object Dictionary).

robot2012, 2022-11-25:
Hi,

Thank you for the response. It makes sense to me. I just wonder if you have any template/example for OD file. I am a newbie in CanOpen development.

Thank you again!
Chun

CANopenNode, 2022-11-29:
dsPIC33 is not updated, for example see PIC32 or https://github.com/CANopenNode/CANopenPIC/pull/8

closed 2023-03-17 by CANopenNode