CANopenNode/CANopenNode #423: How to write to a DOMAIN SDO

andrew-elder, 2023-01-31:
I'm working on a large transfer to a DOMAIN SDO to perform a firmware update. I have left CO_CONFIG_SDO_SRV_BUFFER_SIZE at the default value of 32.

I started by examining the code domainDemo.c code from [here](https://github.com/CANopenNode/CANopenDemo/blob/master/demo/domainDemo.c). But, using the "master" code from [here](https://github.com/CANopenNode/CANopenNode/blob/master/301/CO_SDOserver.c) I keep triggering 

```
                /* if data size exceeds variable size, abort */
                if (SDO->OD_IO.stream.dataLength > 0
                    && SDO->sizeTran > SDO->OD_IO.stream.dataLength
                ) {
                    abortCode = CO_SDO_AB_DATA_LONG;
                    SDO->state = CO_SDO_ST_ABORT;
                    break;
                }
```

from [here](https://github.com/CANopenNode/CANopenNode/blob/master/301/CO_SDOserver.c#L931).

At the start of the transfer `SDO->sizeInd` is set with the transfer size, but then does not seem to be used at all. See [here](https://github.com/CANopenNode/CANopenNode/blob/master/301/CO_SDOserver.c#L880).

The SDO->OD_IO.stream.dataLength appears to be limited to the size of the stream buffer and not set up for filling and then emptying the stream buffer many times. What am I dong incorrectly?

martinwag, 2023-02-01:
Downloading large files is possible using stream mode. Usage of Domain has been asked a few times, search issues for "domain".

CANopenNode, 2023-02-01:
SDO transfer is quite complex and I don't remember details now. Indication of data size is not required by the standard, so it may only be informative and for verification.

Example in CANopenDemo should work without problems. Is it so? If you have own implementation, you may try to do some comparisons.

andrew-elder, 2023-02-01:
I got the CANOpen domain demo to work. I see a sequence that looks like:
```
 (028.225714)  can0  604   [8]  C4 22 21 00 00 00 00 00
 (000.000142)  can0  584   [8]  A4 22 21 00 7F 00 00 00
 (000.000036)  can0  604   [8]  01 00 01 02 03 04 05 06
 (000.000005)  can0  604   [8]  02 07 08 09 0A 0B 0C 0D
 (000.000003)  can0  604   [8]  03 0E 0F 10 11 12 13 14
 (000.000012)  can0  604   [8]  04 15 16 17 18 19 1A 1B
 (000.000003)  can0  604   [8]  05 1C 1D 1E 1F 20 21 22
```
my code has a sequence that looks like
```
    99)      9300.4  Rx         061C  8  21 50 1F 01 E0 12 08 00 
   100)      9303.1  Rx         059C  8  60 50 1F 01 00 00 00 00 
   101)      9305.9  Rx         061C  8  00 E9 07 02 30 A4 15 08 
   102)      9319.7  Rx         059C  8  20 00 00 00 00 00 00 00 
   103)      9322.0  Rx         061C  8  10 40 EE 00 00 00 00 00 
```
so it would seem the cocomm example (that I used for the CANOpen domain demo test) uses SDO Block transfers while my test uses SDO segmented transfers. I'll see if I can switch my implementation to use block transfers.

CANopenNode, 2023-02-02:
Segmented transfer should work too. There may be a bug. Maybe you can reproduce error with segmented transfer and cocomm?
`cocomm "set sdo_block 0"`

CANopenNode, 2023-02-03:
cocomm just sends the string to the canopend. The string is specified by standard CiA309. String does not include information about the variable size, except with known variable types.

If there is a large block of data being transferred, there are multiple reads/writes through socket. In Linux data goes like flow through cocomm (pipes) and data length can't be passed.

CANopenNode, 2023-02-03:
Maybe you can try without data size indication first.

robincornelius, 2023-02-03:
> It is looking more and more like this is my issue. I created this small
> python app that I run against a build of the demo application and it seems
> to work correctly.
>
> #
> # domain_test.py
> #
> # Test block transfers
> #
>
> import argparse
> import os
> import time
> import canopen
>
> FIRMWARE_PATH = './demoDomainTest.bin'
> TRANSFER_SIZE_KB = 16
>
> # Create a test file
> some_bytes = list(range(0, 256))
> some_bytes = some_bytes * (TRANSFER_SIZE_KB * 4)
> immutable_bytes = bytes(some_bytes)
> with open(FIRMWARE_PATH, "wb") as binary_file:
>     binary_file.write(immutable_bytes)
>
> # create object dictionary and domainDemo SDO
> objdict = canopen.objectdictionary.ObjectDictionary()
> member = canopen.objectdictionary.Variable('demoDomain', 0x2122)
> member.data_type = canopen.objectdictionary.DOMAIN
> objdict.add_object(member)
>
> # Connect to the CAN bus
> network = canopen.Network()
> network.connect(bustype='socketcan', channel='can0')
> node = network.add_node(4, objdict)
>
> # open the file
> FILESIZE = os.path.getsize(FIRMWARE_PATH)
> CHUNK_SIZE = 1024
> infile = open(FIRMWARE_PATH, 'rb')
> outfile = node.sdo['demoDomain'].open('wb', size=FILESIZE, block_transfer=False)
>
> # send the file in chunks
> while True:
>     chunk = infile.read(CHUNK_SIZE)
>     if not chunk:
>         break
>     outfile.write(chunk)
>
> # finish
> infile.close()
> network.disconnect()
>
> Thanks for your debug ideas. I'll post back here when I figure it out.

You definitely can send the size indication.  I've got 2 boards both with
CanOpenNode and we were shifting 8000 bytes of data via segmented SDO.
Admittedly this was a read not a write, but when reading the domain the
first packet back had size indication and this worked between the boards
and my canmonitor PC application picked it up and correctly decoded it.

I'm checking my flash loader and it looks like (writing from a C# client to
CanOpenNode device) that i send the filesize, my method is below .,.. you
can see my implementation in all its gory details at
https://github.com/robincornelius/libCanOpenSimple/blob/master/SDO.cs

 //Bigger than 4 bytes we use segmented transfer
   cmd = 0x21;
   expitided = false;
   byte[] payload = new byte[4];
   payload[0] = (byte)databuffer.Length;
   payload[1] = (byte)(databuffer.Length >> 8);
   payload[2] = (byte)(databuffer.Length >> 16);
   payload[3] = (byte)(databuffer.Length >> 24);

            p.cob = (UInt16)(0x600 + node);
            p.len = 8;
            p.data = new byte[8];
            p.data[0] = cmd;
            p.data[1] = (byte)index;
            p.data[2] = (byte)(index >> 8);
            p.data[3] = subindex;

            int sendlength = 4;

            if (payload.Length < 4)
                sendlength = payload.Length;

            for (int x = 0; x < sendlength; x++)
            {
                p.data[4 + x] = payload[x];
            }

            can.SendPacket(p);

then after this is the flip/flop segmented transfer, with the toggle flag
and finally an end flag is set for the last packet

if you have certain canusb devices my canmonitor app will log and decode
your packets and may help with debugging..
https://github.com/robincornelius/CanOpenMonitor

andrew-elder, 2023-02-03:
@robincornelius - thank you for your additional comments. I found a line in my version of the OD_write_domainDemo() implementation that did not match the original. I have fixed it, and now it works correctly.

closed 2023-02-03 by andrew-elder