CANopenNode/CANopenNode #426: SDO Client block transfer has incorrect handling of early server acknowledgments

mpenate-ellenbytech, 2023-02-16:
I am using a self-written CON driver on an MCU without hardware level buffering. I have run into scenarios where overrun has occurred due to calls of `CO_SDOclientDownload` in quick succession. Using a CAN analyzer I can see this:

* Client begins block transfer **with indicated size**
* Client sends packets successfully up to the packet with sequence number `0x29`
* Client sends packet from another CON module, holding up CAN HW
* Client overran and dropped packet with sequence `0x2A`
* Client sends packet with sequence number `0x2B`
* Server sends early acknowledgment response with ack number `0x29`
* Client backs up and starts the sequence numbers over with correct **but does not decrement `SDO_C->sizeTran` by the amount of bytes that it reverted back to**
* Block transfer resumes (sequence number `0x01`) and continues to the end
* Client sees data is still left in SDO fifo but that `SDO_C->sizeTran` indicates that all bytes were sent
* Block transfer fails with abort code `CO_SDO_AB_DATA_LONG` and return value `CO_SDO_RT_endedWithClientAbort`

The incorrect processing is found [here](https://github.com/CANopenNode/CANopenNode/blob/master/301/CO_SDOclient.c#L786) where the SDO client processes received responses from the server

The error was generated [here](https://github.com/CANopenNode/CANopenNode/blob/master/301/CO_SDOclient.c#L1022) where the client tries to generate a new packet to send to the server

Easily reproduced by forcing a single dropped packet in the `CO_CANsend` driver function:

```c
static int forced_drop = 0;
CO_ReturnError_t CO_CANsend(CO_CANmodule_t *CANmodule, CO_CANtx_t *buffer)
{
    //other send code

    //right before hardware send code
    //verified that no other SDO message has 0b0011 as the first nibble in first data byte, 0x30 is safe
    if (ident == CO_SDO_CLIENT_IDENT && data[0] == 0x30 && 0 == forced_drop)
    {
        forced_drop = 1;
    }
    else
    {
        //send message on CAN HW
    }
}
```

I haven't tested or explored the issue on the server side for uploads but I suspect the issue also exists there.

thaanstad, 2023-02-20:
I believe I may be seeing a similiar issue as well.  I haven't done as deep a dive as @mpenate-ellenbytech but I believe the issue is related.  I am working on implementing CANopenNode for an ESP32 processor and when sending SDO download commands from a CAN terminal (600 commands simulating client) I am only able to send the command once where it is recognized.  The second time there is an error in the SDO and I don't receive the 580 response from the server.  Subsequent client commands are not processed and I need to reset the MCU.

I disabled the following block transfer settings in CO_config.h and now SDO commands are processed correctly:

```
#define CO_CONFIG_SDO_SRV_SEGMENTED 0x00
#define CO_CONFIG_SDO_SRV_BLOCK 0x00
```

Still a newbie to CANopenNode so not sure how this will impact other features.  Perhaps I am missing some client-side messages for the block transfer?

mpenate-ellenbytech, 2023-02-20:
@thaanstad I have written a small patch for the library client download code.It covers the first `if` code block in a case statement of the `CO_SDOclientDownload` function, search for the case value.

```c
            case CO_SDO_ST_DOWNLOAD_BLK_SUBBLOCK_REQ:       
            case CO_SDO_ST_DOWNLOAD_BLK_SUBBLOCK_RSP: {
                if (SDO_C->CANrxData[0] == 0xA2) {
                    /* check number of segments */
                    if (SDO_C->CANrxData[1] < SDO_C->block_seqno) {
                        /* NOT all segments transferred successfully.
                         * Re-transmit data after erroneous segment. */
                    	size_t size_before = CO_fifo_altGetOccupied(&SDO_C->bufFifo);
                        CO_fifo_altBegin(&SDO_C->bufFifo,
                                         (size_t)SDO_C->CANrxData[1] * 7);
                        SDO_C->finished = false;
                        size_t size_after = CO_fifo_altGetOccupied(&SDO_C->bufFifo);
                        SDO_C->sizeTran -= size_after-size_before;
                    }
```
I do not know the intricate details of the inner workings of the library and choose the first solution I saw that would work. I am unsure if the double call to `CO_fifo_altGetOccupied` is the best way to calculate the difference in size.

If this patch does not fix your issue, please make a new one.

CANopenNode, 2023-03-17:
I managed to reproduce the error with CANopenLinux according to the above instructions. I also verified the above solution and it is correct, tested and published.

I tested also similar situation on server side for uploads. It works correctly and also seems to be correct [there](https://github.com/CANopenNode/CANopenNode/blob/master/301/CO_SDOserver.c#L1114).

I assume this is fixed. Thank you for the very clear instructions.

CANopenNode, 2023-03-18:
I found better solution, independent from fifo, similar as in SDO server:

    size_t cntFailed = SDO_C->block_seqno - SDO_C->CANrxData[1];
    cntFailed = cntFailed * 7 - SDO_C->block_noData;
    SDO_C->sizeTran -= cntFailed;

Tested and published.

CANopenNode, 2023-03-18:
A comment about transfer reliability:
CAN itself has very high probability that all CAN messages are actually delivered and there are no undetected errors. If error is detected during CAN message transmission, CAN message is re-transmitted by the hardware and application does not even know that.

So if there are errors in block transfer, there is probably something wrong in the driver, like buffer overrun as in the above case. (However, thanks to that error, we found and fixed a bug in SDO client.) SDO block transfer tries to handle some errors, but if everything is correctly implemented, there is very low probability that disturbances even occur.

In [CANopenDemo](https://github.com/CANopenNode/CANopenDemo/tree/master/test) are tests, which can be used to test block transfer, etc on own devices.

closed 2023-03-28 by CANopenNode