CANopenNode/CANopenNode #431: Locking behaviour of CO_error

mlout, 2023-03-15:
We are using CANopenNode revision ca5f357. Our firmware uses FreeRTOS and has multiple tasks which do calls to CO_error via the CO_errorReport and CO_errorReset macros.

Recently we noticed that because we have multiple tasks calling CO_error, we should have a FreeRTOS mutex in the CO_LOCK_EMCY macro, in order to protect the emergency FIFO.

However as we were looking at this, we noticed that when a SDO read of the emergency history happens (register 0x1003 subindex 1), then neither CO_LOCK_EMCY or CO_LOCK_OD is called, although it reads from the same FIFO that CO_error writes to.

What would be the recommend locking behaviour when using a RTOS?
Should we modify the CANopen implementation so that it always locks the object dictionary using CO_LOCK_OD (we are thinking of using the same mutex in both CO_LOCK_EMCY and CO_LOCK_OD) before doing a SDO read?
We could imagine that this issue is not specific to the emergency registers.

mlout, 2023-03-17:
I have opened a PR https://github.com/CANopenNode/CANopenNode/pull/432 to illustrate how we want to fix the issue.

CANopenNode, 2023-03-17:
I agree with you. And would like to discuss a little and re-think CO_LOCK_OD behavior.

In SDO server and SDO client lock is used only, if OD var is PDO/SRDO mappable. This condition was actually my idea and is new inside stack V4. In old versions lock applied always.
Now I think, lock should be applied always, not only if OD var is mappable. Just don't find any benefits, why using the condition.

V4 is rewritten version of the stack and is relative new, so I think, we should just revert CO_LOCK_OD behavior. Also I don't like to add to many configuration options, that part is complex already.

So I suggest just to remove lines with `bool_t lock = OD_mappable(&SDO->OD_IO.stream);` and remove `if (lock)` conditions.

If you agree, just change the PR and I will merge.

mlout, 2023-03-17:
Our reason to make it configurable was to keep the locking minimal for users that do not use multiple tasks/threads accessing the OD. In that case you only have to lock if a OD var is mappable.

I agree in order to keep the code and configuration simple it would be better to always lock. I will change the PR accordingly.

mlout, 2023-03-17:
I have pushed the changes.
OD_mappable is no longer being used by the library, but I decided not to remove it, as it might still be helpful for users who want to check if they should lock in their application code.

CANopenNode, 2023-03-17:
It is merged, Thank you.

closed 2023-03-17 by CANopenNode