CANopenNode/CANopenNode #437: Using more than 2 TPDO

Alrick-gr, 2023-04-07:
Hello,
When I m using more thant 2 TPDOs (configured with the OD_editor), the 2 first TPDOs are sent and then, nothing happens.
In `CO_CANsend` I have the error `CO_ERROR_TX_OVERFLOW`. If i only use 2 TPDOs, it works perfectly.

When i put a breakpoint before `CO_process_TPDO`, then it works and send all TPDOS, but the device will no longer receive any commands.
[image]
[image]
[image]
Here you can see the configuration in the OD_editor. If i check "invalid" for one TPDO it works fine

CANopenNode, 2023-04-11:
Probably there is a problem with the driver.

Wen-Xian-Sen, 2023-05-12:
Hi!I have the same problem.I want to send 5 TPDO, the first TPDO was sent successfully，but the TPDO5 is not send.When i put a breakpoint in `CO_TPDOsend(TPDO)` .It running normal.
`        /* Send PDO by application request or by Event timer */
        if (TPDO->transmissionType >= CO_PDO_TRANSM_TYPE_SYNC_EVENT_LO) {
#if (CO_CONFIG_PDO) & CO_CONFIG_TPDO_TIMERS_ENABLE
            TPDO->inhibitTimer = (TPDO->inhibitTimer > timeDifference_us)
                               ? (TPDO->inhibitTimer - timeDifference_us) : 0;

            /* send TPDO */
            if (TPDO->sendRequest && TPDO->inhibitTimer == 0) {
                CO_TPDOsend(TPDO);
            }

 #if (CO_CONFIG_PDO) & CO_CONFIG_FLAG_TIMERNEXT
            if (TPDO->sendRequest
                && timerNext_us != NULL && *timerNext_us > TPDO->inhibitTimer
            ) {
                /* Schedule for just beyond inhibit window */
                *timerNext_us = TPDO->inhibitTimer;
            }
 #endif
#else
            if (TPDO->sendRequest) {
                CO_TPDOsend(TPDO);
            }
#endif
        } /* if (TPDO->transmissionType >= CO_PDO_TRANSM_TYPE_SYNC_EVENT_LO) */`

Alrick-gr, 2023-05-15:
@Ashenoneq I replied to you on this [issue](https://github.com/CANopenNode/CanOpenSTM32/issues/23)

Alrick-gr, 2023-06-09:
I forgot to update the issue but i solved my problem by changing the size of the heap and stack.
Also i changed the timing in the function CO_process_TPDO.
And now it works perfectly.

closed 2023-06-09 by Alrick-gr