CANopenNode/CANopenNode #446: How can I get the COB-ID in the RPDO callback function?

yamamoto-kazusige, 2023-05-07:
There are nodes with COB-ID 0x38a, 0x38b. Each node sends Position actual value of 6064h with TPDO.

How can I get the COB-ID in the CANopenNode RPDO callback entry?
The callback function is declared as follows and can only get the data body.
```
ODR_t H6064_write(OD_stream_t *stream, const void *buf, OD_size_t count, OD_size_t *countWritten);
```

robincornelius, 2023-05-07:
You don’t. It’s not part of the protocol not a limitation of the library. PDOs do not contain source information, you will have to configure one of your nodes to transmit on a different pod as there is no other way to tell them apart

yamamoto-kazusige, 2023-05-10:
Thank you for your reply. I'll look into the device  settings.

closed 2023-05-10 by yamamoto-kazusige

Mpic, 2023-06-16:
Well, CiA 301 still suggests that the source information (Node ID of the sender) is contained in the TPDO CAN-ID, which would be 180h + Node-ID for TPDO1 for the generic pre-defined connection set.
It would be useful to provide a generic "RPDO" callback with COB-ID to implement PDO reception efficiently for a great number of nodes (without having to define a specific RPDO for each sending node).