CANopenNode/CANopenNode #447: Accessing the OD Index from OD_extension callback function

pivodog, 2023-05-08:
I am trying to find out the OD index of a SDO that is being modified using a call back function specified in the OD_extension_init functions. The only way that that I see is create a callback function for each specified OD index. Is there a better way?

CANopenNode, 2023-11-10:
You can use one callback function for multiple OD entries. Each "extended" OD entry must have own `OD_extension_t` object. Part of it is also a `void *object`, which points to any information you want. That pointer is then passed to callbacks, which are responsible to correctly interpret it.

CANopenNode, 2024-03-13:
That information was really missing inside callback. It is added now: https://github.com/CANopenNode/CANopenNode/commit/1204042a1f83cbbb3dc6398ed410bb9f58fc5102

closed 2024-03-13 by CANopenNode