CANopenNode/CANopenNode #448: [help wantd] CANopenNode 1.3 cannot use the CO_OC.c&h files genratoed by CANopenEditor

AllenMaa, 2023-05-10:
Using CANopenNode v1.3/examples/IO.eds to genrate CO_OC.c&h files.
I have tried using CANopenEditor and libedssharp, but they don't work.

For CANopenEditor v0.8，the generated CO_OD.h file is missing the definiton of COB_ID_TIME.
For CANopenEditor build, the generated CO_OD.h file is missing the definiton of timeOfDay_t.

- CANopenNode v1.3 b92b4bb(https://github.com/CANopenNode/CANopenNode/releases/tag/v1.3)
- CANopenEditor build branch(https://github.com/aerospacesoftware/CANopenEditor/tree/build)
- libedssharp 0.8 Release(https://github.com/robincornelius/libedssharp/releases/tag/v0.8)

AllenMaa, 2023-05-10:
What I also want to know is whether the CO_OD.c&h file under CANopenNode v1.3/examples/ is generated by the old generator? Where can I find this tool?

martinwag, 2023-05-11:
We are using v1.3. The last editor update we did is to [2e53b](https://github.com/robincornelius/libedssharp/commit/2e53bbcb7c012805f02c7b9336efc67c743a95b1). I don't know at which version v1.3 Support was dropped, but this one works for me. You need to compile this from the source.

With this version, you can use insert profile -> DS301 profile.xml. This contains register 0x1012, which when enabled triggers the export for COB_ID_TIME and the missing struct. Be aware that the "enabled" feature (and some more) works only if you save with "file -> export device file -> "CANopenNode XML" in this old version.

AllenMaa, 2023-05-17:
> We are using v1.3. The last editor update we did is to [2e53b](https://github.com/robincornelius/libedssharp/commit/2e53bbcb7c012805f02c7b9336efc67c743a95b1). I don't know at which version v1.3 [...]

Thank you sooo much, it works in using _project.xml file.

AllenMaa, 2023-05-17:
Why don't  consider supporting v1.3 in CANopenEditor as well? After all, v1.3 is the latest version in Release.
@trojanobelix @CANopenNode

trojanobelix, 2023-05-22:
I myself only use V1.3. I am not aware of any problems with the CANopenEditor when using V1.3.

Sure that you using the actual editor from https://github.com/CANopenNode/CANopenEditor?

AllenMaa, 2023-05-24:
I'm sorry, I use CANopenEditor is from https://github.com/aerospacesoftware/CANopenEditor/tree/build. 

When I try to use https://github.com/CANopenNode/CANopenEditor/tree/4.0-104-g6f50f73 to open the _project. XML and insert DS301_profile_old to generatethe CO_OD.c/h file, but the compiler still fails to compile successfully, showing a missing definition of timeOfDay_t. 

By the way, the CO_TIME.h file in v1.3 release version (b92b4bb) does not define timeOfDay_t. 
Perhaps you are using v1.3 c49f119?

AllenMaa, 2023-05-24:
> I myself only use V1.3. I am not aware of any problems with the CANopenEditor when using V1.3.
> 
> Sure that you using the actual editor from https://github.com/CANopenNode/CANopenEditor?
> 
> I do not use the CO_NO_TIME feature, but as far as I see comes the timeOfDay_t from CANopenNode:
> 
> CO_TIME.h #ifndef timeOfDay_t typedef union { unsigned long long ullValue; struct { unsigned long ms:28; unsigned reserved:4; unsigned days:16; unsigned reserved2:16; }; }timeOfDay_t; #endif`
> 
> The COB-ID TIME comes with Object 0x1012, when using **DS_301_old** profile. (in the newer Profile it is named "COB-ID time stamp object")

I'm sorry, I use CANopenEditor is from https://github.com/aerospacesoftware/CANopenEditor/tree/build. 

When I try to use https://github.com/CANopenNode/CANopenEditor/tree/4.0-104-g6f50f73 to open the _project.xml and insert DS301_profile_old.xml to generatethe CO_OD.c/h files, but the compiler still fails to compile successfully, showing a missing definition of timeOfDay_t. 

By the way, the CO_TIME.h file in v1.3 release version (b92b4bb) does not define timeOfDay_t. 
Perhaps you are using v1.3 c49f119?