CANopenNode/CANopenNode #449: [EMCY] Question on error status bits

samsamfire, 2023-05-10:
Hello,

I've been scrolling through CiA specifications, specifically on the emergency protocol.
Maybe I have missed something but apart from the CANopenNode documentation I can't find anything related to the "errorBit" and errorStatusBits online.
The last version of CiA 301 shows that the "manufacturer specific" information should be exactly 5 bytes long.
[image]

Does that mean that the stack "intends" to have manufacturer specific information ?

Thanks in advance !

martinwag, 2023-05-11:
Yes the stack has manufacturer specific info. One byte for general error and 4 bytes for details. You can also use them as described here: [https://github.com/CANopenNode/CANopenNode/blob/361080da879b308876d587c60dbcc93d14104bf9/301/CO_Emergency.h#LL614C6-L614C14](https://github.com/CANopenNode/CANopenNode/blob/361080da879b308876d587c60dbcc93d14104bf9/301/CO_Emergency.h#LL614C6-L614C14)

samsamfire, 2023-05-12:
Thank you martin !

closed 2023-05-12 by samsamfire