CANopenNode/CANopenNode #452: How to set the STM as a NMT master and send NMT commands?

TimK456, 2023-05-15:
Hi,

I have import the CANopenNodeSTM32 to my project and configurate the heartbeat and PDOs via CANopenEditor. Thats works fine. My second device is a slave and starts in the pre-operational mode. So I would send the NMT command from the STM to set the second device to the operational mode.

My question is now, how to set the STM to a NMT master and how can I send the NMT commands like the operational mode. I think it should be also possible to configurate it in the editor and send it.

martinwag, 2023-05-17:
You can set the default values in the editor. If I remember correctly, you should get a define CO_NO_NMT_MASTER in CO_OD.h when OD 0x1f80 bit 0 is set.

elotom, 2023-05-23:
> I have also a problem changing the NMT master to the operational state. I'm using CO_NMT_sendCommand to enter operational with a node id = 0 which means that all nodes including self should enter operational mode. I have also added OD 0x1F80 but I don't know what value should be set 0x1F80 because of the lack of the description. [image]

It was a driver issue.

TimK456, 2023-05-24:
I insert the NMTStartup and set the bit 0. So I get x1F80_NMTStartup in the OD.h file, but no CO_NO_NMT_MASTER is defined and it still not working. I also try to set the bit 3 instead of bit 0 to start in the pre-operational mode (just for testing), but it starts continue in the operational mode.

[image]
This are my settings of the object.

spaceKelan, 2023-06-19:
Hi @TimK456 - the 0x1f80 register is described in CiA 302. You can find a draft version [here](http://overpof.free.fr/schneider/CAN%20&%20CANopen/CANopen/%A9CiA%20CANCANopen%20CD%20V5.1/standard/dsp30232.pdf) chapter 5.1 is relevant. I don't understand yet, how this is dealt with in CANopenNode, which is why I just created [an issue](https://github.com/CANopenNode/CANopenNode/issues/464)

CANopenNode, 2023-11-10:
0x1f80 is used by v1.3, but not in V4.0. It can be added manually, if necessary. See `NMT_CONTROL`. 

Use `CO_CONFIG_NMT_MASTER` to enable `CO_NMT_sendCommand` function.

closed 2023-11-14 by TimK456