CANopenNode/CANopenNode #454: Get app notification on SDO read/write

MaJerle, 2023-05-19:
@CANopenNode , what is the most effective way to get notification on SDO write/read from the client?

Use case - Device has DAC outputs that can be configured through SDO protocol (device acts as server). How can application on server device receive callback that read/write for specific index happened?

I'd like to update the external DAC device settings, when this happens.

Today the solution is to keep copy of SDO values in another array and periodically check if any value in SDO got changed.

CANopenNode, 2023-05-20:
Each read or write from object dictionary (SDO or PDO) goes through callback. Default callbacks simply reads or writes defined memory location. But user application may define own callbacks, where it can do own processing.
See:

- https://github.com/CANopenNode/CANopenDemo/blob/master/demo/objectAccessOD.c
- https://canopennode.github.io/CANopenNode/group__CO__ODinterface.html

MaJerle, 2023-05-20:
Would you accept the pull-request if a *generic callback event system* is added to the library?

In that case, an application could register itself and could receive events from different canopennode points.

For instance, some of the events lib could send to app (and let app to do whatever they want with that info).

* SDO write event occurred on index & subindex (index and subindex info are passed as parameter to event)
* SYNC event received
* Heart-beat event received (or sent too).

Etc..

Something like that with LwEVT lib:

* Consumer - https://github.com/MaJerle/lwevt/blob/develop/examples/example_basic_consumer.c
* Producer - https://github.com/MaJerle/lwevt/blob/develop/examples/example_basic_producer.c