CANopenNode/CANopenNode #460: Master what is your Node Id?

david-oz00, 2023-06-07:
Hi, can anyone please confirm this observation:

it seems CANopenNode requires the bus master to have an assigned Node Id and it can't be zero.  This is true even if the LSS and SDO servers aren't implemented.  Consequently, I must ensure all slave devices conform to this:  (2<=  slave node id <= 127)......since the bus master will use node id=1.

Cheers
David

martinwag, 2023-06-07:
Yes, every device needs a node ID, as otherwise you can't conform to specification (you can't have the mandatory objects).

david-oz00, 2023-06-07:
Fair enough.  Some CANopen documentation state it supports up to 127 slave devices (node ids 1..127), but it must be including the master in that mix.  Thanks for the clarification.  I'll close the question.

closed 2023-06-07 by david-oz00

martinwag, 2023-06-07:
on this level, there is no differentiation between master and slave devices within CANopen. A slave becomes master by adding master functionality. You can even distribute master functionality between multiple devices (e.g. NMT master on one and LSS master on another device)