CANopenNode/CANopenNode #461: How to get subIndex?

Ashenoneq, 2023-06-12:
`OD_set_value(OD_ENTRY_H2006, 0, sdata, 4, false);`
This function requires an entry and a subIndex. But the dictionary editor doesn't have macros to generate subindexes.How can i get the subindex values?

martinwag, 2023-06-12:
Does your OD entry have members? if not, no subindex is generated and you can pass 0 to the function.

Ashenoneq, 2023-06-13:
> Does your OD entry have members? if not, no subindex is generated and you can pass 0 to the function.

`OD_ATTR_RAM OD_RAM_t OD_RAM = {
    .x1001_errorRegister = 0x00,
    .x1010_storeParameters_sub0 = 0x04,
    .x1010_storeParameters = {0x00000001, 0x00000001, 0x00000001, 0x00000001},
    .x1011_restoreDefaultParameters_sub0 = 0x04,
    .x1011_restoreDefaultParameters = {0x00000001, 0x00000001, 0x00000001, 0x00000001},
    .x1200_SDOServerParameter = {
        .highestSub_indexSupported = 0x02,
        .COB_IDClientToServerRx = 0x00000600,
        .COB_IDServerToClientTx = 0x00000580
    },
    .x200A_TTT = {
        .highestSub_indexSupported = 0x02,
        .subObject1 = 0x00000000,
        .subObject2 = 0x00000000
    },
    .x6000_PMM_counter = 0x00000000
};`
This is my OD.c. I want to use `OD_set_value` to set  the x200A_TTT's subObject1 or subObject2, but I don't want to go through the input 0,1,2......
Is there an easy way to get the subIndex? I find the OD.h have many macro such as `define OD_ENTRY_H1000`,but i did't find any macro define about subIndex.Do you have any ideas?

martinwag, 2023-06-13:
you should have something like this for each member in CO_OD.h
```
/*6200 */
        #define OD_6200_writeOutput8Bit                             0x6200

        #define OD_6200_0_writeOutput8Bit_maxSubIndex               0
        #define OD_6200_1_writeOutput8Bit_output07                  1
```
containing OD index, member count and subindex for each member.

Ashenoneq, 2023-06-13:
> you should have something like this for each member in CO_OD.h
>
> ```
> /*6200 */
>         #define OD_6200_writeOutput8Bit                             0x6200
>
>         #define [...]

Hi! Is CO_OD.h a file you defined yourself? I realized I didn't have this file

martinwag, 2023-06-13:
You need to use the OD editor
https://github.com/CANopenNode#object-dictionary-editor

Ashenoneq, 2023-06-14:
> You need to use the OD editor https://github.com/CANopenNode#object-dictionary-editor
My OD file is generated using OD editor, but the filename is not CO_OD.h, is the version we use different?

martinwag, 2023-06-14:
I've checked with the lastest version, indeed this uses different file name and the defines are gone. Sorry, can't help you then. @CANopenNode , how is this supposed to work?

closed 2023-07-15 by Ashenoneq