CANopenNode/CANopenNode #467: How do I cancel the first send after TPDO startup?

Ashenoneq, 2023-08-04:
When my device switches to the operation state, TPDO will automatically send once, but I don't want it to. How can I cancel the first send?

martinwag, 2023-08-04:
You can disable a TPDO by default in the TPDO communication parameters, and then enable it when you are ready for reception.

Ashenoneq, 2023-08-04:
Thanks, I tried that, but when I enable TPDO, they still send it once

martinwag, 2023-08-04:
That's the intended behavior. How is the device supposed to know when to start sending PDOs? A fast changing input could be triggering TPDOs e.g. every 1ms, so you wouldn't even know the first one was dropped.
If you want to control TPDO transmission, you need to use sync-pdo.

Ashenoneq, 2023-08-04:
> That's the intended behavior. How is the device supposed to know when to start sending PDOs? A fast changing input could be triggering TPDOs e.g. every 1ms, so you wouldn't even know the first one was dropped. If you want to control TPDO transmission, you need to use sync-pdo.

My TPDO send is asynchronous:
`
    .x181E_TPDOCommunicationParameter = {
        .highestSub_indexSupported = 0x06,
        .COB_IDUsedByTPDO = 0x0000019F,
        .transmissionType = 0xFE,
        .inhibitTime = 0x03E8,
        .eventTimer = 0x00,
        .SYNCStartValue = 0x00
    },
`
If my program doesn't control it to send, it won't send.

However, after the protocol stack is running, TPDO is sent once after switching from the pre-operation state to the operation state.

I don't quite understand the reason for the design. I guess it is the first time to send all of them, so that the receiver can confirm whether the TPDO of the sender is normal, is that right?

martinwag, 2023-08-04:
OK, now I understand, you want to control PDO transmission from your own software instead of using integrated functions. I've had a look in the CO_PDO.h, there is no method to trigger a manual PDO transmit, did you do your own extension to support that?

Ashenoneq, 2023-08-05:
Yes, I used the OD_extension_init function. 
I used asynchronous sending in my code, calling the OD_requestTPDO function

martinwag, 2023-08-07:
If I understand the docs correctly, this only requests an additional send of a TPDO. The TPDO is then sent in the PDO processing loop. The other available Events (start / value change) can also trigger a TPDO send, which is the behavior you observe.

In v1.3 Stack this was the same, +there was no method available for PDO processing by user. From what I've seen now this still seems to be the case for v4. If you need this functionality you have to modify the stack code.

Ashenoneq, 2023-08-07:
> If I understand the docs correctly, this only requests an additional send of a TPDO. The TPDO is then sent in the PDO processing loop. The other available Events (start / value change) can also trigger a TPDO send, which is the behavior you observe.
> 
> In v1.3 Stack this was the same, there was no method available for direct PDO processing by user. From what I've seen now this still seems to be the case for v4. If you need this functionality you have to modify the stack code.

Thank you for your reply!

CANopenNode, 2023-08-08:
A little extra explanation:

> When my device switches to the operation state, TPDO will automatically send once, but I don't want it to. How can I cancel the first send?

By default it is sent as soon TPDO is enabled (depends on transmission type). Then according to event timer or application request. For more info search for `TPDO->sendRequest` inside CO_PDO.c file.

The idea is, data sent by TPDO is always up to date, not counting messages. CAN module may also send a message twice, if there are disturbances on the bus.

Ashenoneq, 2023-08-09:
> A little extra explanation:
>
>
> By default it is sent as soon TPDO is enabled (depends on transmission type). Then according to event timer or application request. For more info search for [...]
Yes, my TPDO transmission type is 254(Asynchronous events). 

I noticed that every time I started the protocol stack, the `CO_TPDO_init` would set `TPDO->sendRequest` to true and then send it once.

The reason why I don't want to send it for the first time is that I have many devices, and when they are powered on at the same time, they may occupy the bus at the same time and increase the bus load

CANopenNode, 2023-08-09:
> The reason why I don't want to send it for the first time is that I have many devices, and when they are powered on at the same time, they may occupy the bus at the same time and increase the bus load

CAN should handle that. Or there is a problem?
If increased bus load is a problem, then the same may happen in case of some disturbance in the system, where change of state may be detected on many nodes simultaneously. In that case you may use more deterministic approach with synchronous PDOs.

You could use NMT master and start nodes in sequence.

OR you could simply comment some lines with `TPDO->sendRequest = true;`

Ashenoneq, 2023-08-10:
Thank you for your suggestion, I will give it a try.

closed 2023-08-10 by Ashenoneq

BZU2020, 2024-01-02:
> 是的，我使用了 OD_extension_init 函数。 我在代码中使用异步发送，调用 OD_requestTPDO 函数

You call the function OD_ RequestTPDO, how to use OD_ Can requestTPDO be used to send TPDO?