CANopenNode/CANopenNode #468: SDO client download example

jordibdlll, 2023-08-10:
Hi,

Looking at the **example code** you provide in "**CO_SDOclient.h**", in CO_SDO_abortCode_t **write_SDO**(CO_SDOclient_t *SDO_C, uint8_t nodeId, uint16_t index, uint8_t subIndex,uint8_t *data, size_t dataSize), i have realized that if the SDO buffer is full you remain in the loop for ever. 

I became to this situation while trying to send a message in a CAN bus with **only one device** (the embedded) for testing purposes. If only one node is connected (with no monitors such as PCAN Basic / Kvaser / candump ...), when the embedded starts up tries to send the "BOOT" but no one receives it so it will send "EMCY" messages too. At some point the **tx buffer** will be **full** and the code will remain stucked at:
`    do {
        uint32_t timeDifference_us = 10000;
        CO_SDO_abortCode_t abortCode = CO_SDO_AB_NONE;

        SDO_ret = CO_SDOclientDownload(SDO_C,
                                       timeDifference_us,
                                       false,
                                       bufferPartial,
                                       &abortCode,
                                       NULL, NULL);
        if (SDO_ret < 0) {
            return abortCode;
        }

        sleep_us(timeDifference_us);
    } while(SDO_ret > 0);`
    
    Note that CO_SDO_RT_transmittBufferFull = 4, so the function does not return. Could you tell me why is this designed as it is? Is there any way to not block the code under the commented conditions?
    
    Thank you

Ashenoneq, 2023-08-14:
Hello! Our device has a high requirement for real-time performance and cannot be blocked for too long, so I have the same concern. Currently, my approach is to have the caller poll this function multiple times until it returns either success or failure before exiting.

`extern SdoAbortCode writeSdoClientNoBlock(uint8_t nodeId, uint16_t index, \         `
`                                          uint8_t subIndex, uint8_t* data,\         `
`                                          size_t dataSize)                          `
`{                                                                                   `
`    static uint8_t  first_entry = 0;                                                `
`                                                                                    `
`    static bool_t bufferPartial = false;                                            `
`                                                                                    `
`    CO_SDO_return_t SDO_ret;                                                        `
`    CO_SDOclient_t* SDO_C = CO->SDOclient;                                          `
`                                                                                    `
`    if (first_entry == 0)                                                           `
`    {                                                                               `
`        first_entry = 1;                                                            `
`                                                                                    `
`        // setup client (this can be skipped, if remote device is the same)         `
`        SDO_ret = CO_SDOclient_setup(SDO_C,                                         `
`                                     CO_CAN_ID_SDO_CLI + nodeId,                    `
`                                     CO_CAN_ID_SDO_SRV + nodeId,                    `
`                                     nodeId);                                       `
`        if (SDO_ret != CO_SDO_RT_ok_communicationEnd)                               `
`        {                                                                           `
`            return USER_ERR_CODE;                                                   `
`        }                                                                           `
`                                                                                    `
`        // initiate download                                                        `
`        SDO_ret = CO_SDOclientDownloadInitiate(SDO_C, index, subIndex,              `
`                                               dataSize, 1000, false);              `
`        if (SDO_ret != CO_SDO_RT_ok_communicationEnd)                               `
`        {                                                                           `
`            return USER_ERR_CODE;                                                   `
`        }                                                                           `
`                                                                                    `
`        // fill data                                                                `
`        size_t nWritten = CO_SDOclientDownloadBufWrite(SDO_C, data, dataSize);      `
`        if (nWritten < dataSize)                                                    `
`        {                                                                           `
`            bufferPartial = true;                                                   `
`            // If SDO Fifo buffer is too small, data can be refilled in the loop.   `
`        }                                                                           `
`    }                                                                               `
`                                                                                    `
`    //download data                                                                 `
`    CO_SDO_abortCode_t abortCode = CO_SDO_AB_NONE;                                  `
`    uint32_t timeDifference_us = 1000;                                              `
`    SDO_ret = CO_SDOclientDownload(SDO_C,                                           `
`                                   timeDifference_us,                               `
`                                   false,                                           `
`                                   bufferPartial,                                   `
`                                   &abortCode,                                      `
`                                   NULL, NULL);                                     `
`                                                                                    `
`    if (SDO_ret == 0) //success                                                     `
`    {                                                                               `
`        first_entry = 0;                                                            `
`        bufferPartial = false;                                                      `
`        return REDEF_CO_SDO_AB_NONE;                                                `
`    }                                                                               `
`    else if (SDO_ret > 0)                                                           `
`    {                                                                               `
[... cut ...]
`    {                                                                               `
`        first_entry = 0;                                                            `
`        bufferPartial = false;                                                      `
`        return (SdoAbortCode)abortCode;                                             `
`    }                                                                               `
`}                                                                                   `

CANopenNode, 2023-11-10:
SDO client is quite complex and has to be implemented properly and well integrated into OS. In CANopenLinux it works really fast, no delays, very reliable. See there and also CANopenDemo.