CANopenNode/CANopenNode #469: Multiple CANopen Networks on Single Board

awolf-illumina, 2023-08-21:
Is it possible to run multiple CANopenNode networks on a single micrcontroller?

CANopenNode, 2023-11-10:
Yes, it is object oriented. But may be complex.

NieLuc, 2024-08-19:
I am having issues with a too high dataload. The 1 MBit Baud isnt sufficient and therefore i have to split the CANopen Bus into two. Can I just add another transceiver and use the same OD for both busses?

CANopenNode, 2024-08-19:
Yes. Each CANopen object has reference to can module. I used some PDOs on second CAN module. You should adapt CANopen.h and .c file.

NieLuc, 2024-08-20:
Okay, so right now everywhere where only CO->CANmodule[0] is used I should implement a CO->CANmodule[1] with the same functionality if I understand that correctly?

CANopenNode, 2024-08-20:
For example, CO_RPDO_init(), you have to specify your other CANdevRx and CANdevRxIdx (index inside your other CANdev).

You have to call CO_CANmodule_init() to initialize your other CANdev first.

NieLuc, 2024-08-22:
I had to take over a project for Formula Student where we need to use CANopen on a ESP32. 

ecu_state_t initCanOpen()
{
    CO_ReturnError_t err;
    uint32_t heapMemoryUsed;

    CO_CANsetPins(CAN_RX, CAN_TX);

    err = CO_new(&heapMemoryUsed);

    if (err != CO_ERROR_NO)
    {
        debugLog(LOG_LEVEL_ERROR, "Can't allocate memory");
    }
    else
    {
        debugLog(LOG_LEVEL_VERBOSE, 
                 "Allocated %ld bytes for objects", 
                 heapMemoryUsed);
    }

    return ECU_STATE_CANOPEN_RESET;
}

ecu_state_t canOpenResetCommunication()
{
    CO_ReturnError_t err;
    void*            CANmoduleAddress = NULL;
    uint8_t          pendingNodeId    = 1;
    uint8_t          activeNodeId     = 1;
    uint16_t         pendingBitRate   = 500;

    if (CAN_JOK)
    {
        debugLog(LOG_LEVEL_WARNING, "Attempting to un-jok CAN");
    }

    const esp_timer_create_args_t periodic_timer_args = { .callback = &coRealtimeTask, 
                                                          .name = "co_realtime" };

    debugLog(LOG_LEVEL_INFO, "Reset communication");
    CANopenConfiguredOK = false;

    err = CO_CANinit(CANmoduleAddress, pendingBitRate);
    if (err != CO_ERROR_NO)
    {
        debugLog(LOG_LEVEL_ERROR, 
                 "CAN initialisation failed: %d", 
                 err);
    }

    err = CO_LSSinit(&pendingNodeId, &pendingBitRate);
    if (err != CO_ERROR_NO)
    {
        debugLog(LOG_LEVEL_WARNING, 
                 "LSS initialisation failed: %d", 
                 err);
    }

    activeNodeId = pendingNodeId;
    err = CO_CANopenInit(activeNodeId);

    if (err == CO_ERROR_NO)
    {
        debugLog(LOG_LEVEL_INFO, "Initialisation successful");
        CANopenConfiguredOK = true;
    }
    else if (err != CO_ERROR_NODE_ID_UNCONFIGURED_LSS)
    {
        debugLog(LOG_LEVEL_ERROR,
                 "CANopen Initialisation failed: %d",
                 err);
    }

    if (!CAN_JOK)
    {
        esp_timer_handle_t periodicTimer;
        ESP_ERROR_CHECK(esp_timer_create(&periodic_timer_args, &periodicTimer));
        ESP_ERROR_CHECK(esp_timer_start_periodic(periodicTimer, 1000));
    }

    if (CANopenConfiguredOK)
    {
        CO_CANsetNormalMode(CO->CANmodule[0]);
        CO_EM_initCallbackRx(CO->em, &heartbeatCallback);

        const int8_t idx = CO_HBconsumer_getIdxByNodeId(CO->HBcons, NODE_ID_THROTTLE_SENSOR_LEFT);
        CO_HBconsumer_initCallbackTimeout(CO->HBcons, idx, NULL, &hbTimeoutCallbackForThrottleSensors);
        const int8_t idx2 = CO_HBconsumer_getIdxByNodeId(CO->HBcons, NODE_ID_THROTTLE_SENSOR_RIGHT);
        CO_HBconsumer_initCallbackTimeout(CO->HBcons, idx2, NULL, &hbTimeoutCallbackForThrottleSensors);

        return ECU_STATE_START_TASKS;
    }

    return ECU_STATE_ERROR;
}
So if we take the first function that is being used which is `err = CO_new(&heapMemoryUsed` and go to the implementation of that I see 
```
/* Create objects from heap ***************************************************/
#ifndef CO_USE_GLOBALS
CO_ReturnError_t CO_new(uint32_t *heapMemoryUsed)
{
    int16_t i;
    uint16_t errCnt = 0;
    uint32_t CO_memoryUsed = 0;

    /* If CANopen was initialized before, return. */
    if (CO != NULL)
    {
        return CO_ERROR_NO;
    }

    /* globals */
    CO = &COO;

    /* CANmodule */
    CO->CANmodule[0] = (CO_CANmodule_t *)calloc(1, sizeof(CO_CANmodule_t));
    if (CO->CANmodule[0] == NULL)
        errCnt++;
    CO_CANmodule_rxArray0 =
        (CO_CANrx_t *)calloc(CO_RXCAN_NO_MSGS, sizeof(CO_CANrx_t));
    if (CO_CANmodule_rxArray0 == NULL)
        errCnt++;
    CO_CANmodule_txArray0 =
        (CO_CANtx_t *)calloc(CO_TXCAN_NO_MSGS, sizeof(CO_CANtx_t));
    if (CO_CANmodule_txArray0 == NULL)
        errCnt++;
    CO_memoryUsed += sizeof(CO_CANmodule_t) +
                     sizeof(CO_CANrx_t) * CO_RXCAN_NO_MSGS +
                     sizeof(CO_CANtx_t) * CO_TXCAN_NO_MSGS;
[...]
```
With my knowledge right now I would do the same with CANmodule[1]:

```
    /* CANmodule 1*/
    CO->CANmodule[0] = (CO_CANmodule_t *)calloc(1, sizeof(CO_CANmodule_t));
    if (CO->CANmodule[0] == NULL)
        errCnt++;
    CO_CANmodule_rxArray0 =
        (CO_CANrx_t *)calloc(CO_RXCAN_NO_MSGS, sizeof(CO_CANrx_t));
    if (CO_CANmodule_rxArray0 == NULL)
        errCnt++;
    CO_CANmodule_txArray0 =
        (CO_CANtx_t *)calloc(CO_TXCAN_NO_MSGS, sizeof(CO_CANtx_t));
    if (CO_CANmodule_txArray0 == NULL)
        errCnt++;
    CO_memoryUsed += sizeof(CO_CANmodule_t) +
                     sizeof(CO_CANrx_t) * CO_RXCAN_NO_MSGS +
                     sizeof(CO_CANtx_t) * CO_TXCAN_NO_MSGS;

    /* CANmodule 2*/
    CO->CANmodule[1] = (CO_CANmodule_t *)calloc(1, sizeof(CO_CANmodule_t));
    if (CO->CANmodule[1] == NULL)
        errCnt++;
    CO_CANmodule_rxArray1 =
        (CO_CANrx_t *)calloc(CO_RXCAN_NO_MSGS, sizeof(CO_CANrx_t));
    if (CO_CANmodule_rxArray1 == NULL)
        errCnt++;
    CO_CANmodule_txArray1 =
        (CO_CANtx_t *)calloc(CO_TXCAN_NO_MSGS, sizeof(CO_CANtx_t));
    if (CO_CANmodule_txArray1 == NULL)
        errCnt++;
    CO_memoryUsed += sizeof(CO_CANmodule_t) +
                     sizeof(CO_CANrx_t) * CO_RXCAN_NO_MSGS +
                     sizeof(CO_CANtx_t) * CO_TXCAN_NO_MSGS;
```

If I look at the next function `CO_CANinit` then inside of it the CO_CANmodule_init function gets `CO->CANmodule[0]` as a parameter. Must I call the CO_CANmodule_init function again with `CO->CANmodule[1]` as a parameter?

CANopenNode, 2024-08-30:
Sorry for my late answer. Files CANopen.h and .c offers mush flexibility, so they seems complex.

However, you are right. You have to allocate memory for CO_CANmodule_rxArray1, CO_CANmodule_txArray1 and CANmodule1. You don't have to put them into CO. Then you have to CO_CANmodule_init with those objects. 

Then you may use your second CANmodule for initialization of any CANopen object. For example, if you have 6 TPDOs and you want to use TPDO 5 and 6 with second module:

- edit CO_CANopenInitPDO
- Make CO_TPDO_init() so, it will use arguments `co->CANmodule, CO_GET_CO(TX_IDX_TPDO) + i` from second module.

Please note, CANopen.h and .c files are for general usage. You can do some "magic" there, but if you want to have full control, you can create them completely new according to your requirements.

If you have problems with bus load, you may also try different strategies with single CAN module. You can get good results, if you use SYNC with PDOs. And make sure, you driver works correctly. For example, CANopenPIC example has no problems of full PDO communication without delays inside 1ms timer task. Also CANopenLinux, if it is used in RT mode. I'm not sure, how driver for ESP32 works.

ragnarmoberg, 2024-11-29:
Im about to do the same type of modification. For my use case the device recieves data in some networks and send diferent but related data on others. Would it be best to do this using a single OD with multiple devices or separate them?

CANopenNode, 2024-11-29:
The question is not very clear.

If you only need some PDOs to work on different networks, you can only configure those PDOs (init PDO with alternate CO_CANmodule_rxArray1, CO_CANmodule_txArray1 and CANmodule1). All other can stay the same, single canopen device, single OD.