CANopenNode/CANopenNode #473: Example to exclude OD.h from CANopen.c

sam-tj, 2023-09-13:
Great work on the repositories :+1:  Really appreciate the work done by you and all the contributors.

I had a small issue importing my OD.h to a build for ESP32 (https://github.com/thaanstad/ESP32_CanOpenNode). 

The repo mentioned above is built using the source code and here https://github.com/CANopenNode/CANopenNode/blob/361080da879b308876d587c60dbcc93d14104bf9/CANopen.c#L35 it imports "OD.h". And in changelog it mentions it can be optional (https://github.com/CANopenNode/CANopenNode/blob/master/doc/CHANGELOG.md#changed)

The question is how to import my **OD.h** in **CANopen.c** such that it gets the same info as my main program without overwriting the original that is provided with the library.

Thanks in advance.

CANopenNode, 2023-11-10:
You have to setup include paths for your project. Don't know how to do it for ESP.

Originally OD.h and OD.c files are in `example` folder. Use own folder for your application and exclude `example`. See Makefile.

sam-tj, 2023-12-03:
Hi @sicrisembay ,
Thanks for the reply. I am using the arduino ide based code. The link is in my question for reference. As far as I know, there is no cmakelist for the arduino framework. If it is included in the .ino file, then it compiles. But as the original OD is still in the library, it uses that.

SWolfSchunk, 2026-08-19:
Maybe it would be a good idea to introduce a separate source folder for sources.
Instead of including into a cmake project for example like this:
```
include_directories("../submodules/CANopen/CANopenNode")
file(GLOB_RECURSE SOURCES "../submodules/CANopen/CANopenNode/*.*")
list(FILTER SOURCES EXCLUDE REGEX "../submodules/CANopen/CANopenNode/example/")
```
it could be done just like this:
```
include_directories("../submodules/CANopen/CANopenNode/inc")
file(GLOB_RECURSE SOURCES "../submodules/CANopen/CANopenNode/src/*.*")
```
This way you will not accidently include or compile example code that may get very confusing.