CANopenNode/CANopenNode #476: Wrong TPDO initialization sequence in case of dynamic OD

itxs, 2023-10-19:
In CO_PDO.c in function `CO_TPDO_init()` PDO object extension initialization functions:
```
OD_extension_init(OD_18xx_TPDOCommPar, &PDO->OD_communicationParam_ext);
OD_extension_init(OD_1Axx_TPDOMapPar, &PDO->OD_mappingParam_extension);
```
are called after
`PDO_initMapping(PDO, OD, OD_1Axx_TPDOMapPar, false, errInfo, &erroneousMap);`
which has checking inside:
```
if (subIndex < (OD_FLAGS_PDO_SIZE * 8) && entry->extension != NULL) {
    PDO->flagPDObyte[mapIndex] =
            &entry->extension->flagsPDO[subIndex >> 3];
    PDO->flagPDObitmask[mapIndex] = 1 << (subIndex & 0x07);
}
else {
    PDO->flagPDObyte[mapIndex] = NULL;
}
```
which is always skipped.

The PDO_initMapping() must be called after OD_extension_init() calls, not before.

CANopenNode, 2023-10-19:
Actually this is not really a problem.

1. `OD_extension_init()` initializes OD entries 0x18xx and 0x1Axx.
2. But `entry` from `PDOconfigMap()` points to any OD entry, which is mappable and is actually mapped to specific PDO.

OD entries 0x18xx and 0x1Axx are never mappable, so they are never subject to checking `entry->extension != NULL`.

`CO_TPDO_init()` is called from `CO_CANopenInitPDO()`, which is called after all other CANopen objects are configured.

However, I didn't test `flagPDO` functionality very widely, so there may still be problems.

itxs, 2023-10-23:
Ah, I see. I can close the issue, thank you for clarifying

closed 2023-10-23 by itxs