CANopenNode/CANopenNode #479: How to port CANopenNodel to TMS320F28375D

WeiShengxiujing, 2023-11-15:
Hello, I would like to ask you some questions. Based on your case, I have ported CANopenNode to F28375D, which can send heartbeat and PDO. However, I have a serious problem: my CPUTimer1 is 1ms, Heartbeat and PDO settings are sent once a second, but the sending speed is very fast, never 1s.

I can ensure that the heartbeat messages and timer events in PDO I set are set every 1s, as I am proficient in configuring and using CANopenNode's heartbeat messages, PDO, and SDO on STM32. After setting up the heartbeat message and PDO, 375D will send me a message in approximately 30-100ms, which I have never seen before on STM32。

The attachment contains my preliminary ported project, using CCS10.0.0 and TI18.0 as the compilation chain. The library function uses driverlib. The attachment contains settings for the heartbeat message I received and screenshots of the received message. You can watch it。

I followed the migration steps, first configuring the sending and receiving interrupts of the CANB bus, then configuring the CPUTimer1 timer bit 1ms interrupt, and then initializing the CANopenNode. I added the CANopenNode function to the CPUTimer1 interrupt and CANBISR, and also added it to while1,

The possible problem is on the CPUTimer1 timer, but I haven't found out exactly where the problem is.

Thank you very much for taking the time to reply to me.
[F28375d_CAN_CANopenNode.zip](https://github.com/CANopenNode/CANopenNode/files/13362070/F28375d_CAN_CANopenNode.zip)

CANopenNode, 2023-11-18:
Porting is not easy. 
You can check timeDifference_us argument in CO_process(), etc.

WeiShengxiujing, 2023-11-20:
I'll give it a try，Thank you very much for taking the time to reply to me.