CANopenNode/CANopenNode #480: LSS Identify Remote Slave function?

a-leibold, 2023-11-20:
Hi,

I have a STM32 CANopen slave running with CANopenNode.
Now I want to use Identify Remote Slave as shown in the CiA 305 documentation.
As master i use the emotas CANopen DeviceExplorer.
The master sends the following codes.
[image]

Is the Identify Remote Slave fuctionallity implemented?

best regards

CANopenNode, 2023-11-21:
It is, but 1018 object may have to be configured.
For details please see the standard or https://canopennode.github.io/CANopenNode/group__CO__LSS.html#details or https://github.com/CANopenNode/CANopenDemo/blob/master/tutorial/LSS.md

a-leibold, 2023-11-22:
Hi,

thank you for your answer.
But I don´t mean the fastscan, I mean the identify remote slave prozess.
The prozess send:
0x46 with vendor
0x47 with productcode
0x48 with minimal revision boarder
0x49 with maximum revision boarder
0x4a with minimal serial boarder
0x4b with maximum serial boarder

Following and example to find the DEDITEC COS-DIO8
[image]

In the stack I don´t find makros or code to handle this?
Did I missed it?

best regards

martinwag, 2023-11-22:
@CANopenNode i've only implemented lss fastscan in v1.3, was this added in v4?

a-leibold, 2023-11-22:
Hi,

I mean the identifiy remote slave service from the CiA Draft Standard Proposal 305 V1.1.1 from 5. Nov. 2002
Chapter 5.6 Identification Protocols, page 22
The service is also mentioned in CiA Draft Standard Proposal 305 V2.2 from 26. Aug. 2008
I will create a pull request.

best regards

CANopenNode, 2023-11-22:
> @CANopenNode i've only implemented lss fastscan in v1.3, was this added in v4?

Yes, in V4 is the same + some update.