CANopenNode/CANopenNode #481: TIME protocol not sending trace at interval

danirpg, 2023-11-20:
I put the producer TIME to 1000ms and i see the 0x100 sends at not exactly 1000ms.
I see the function CO_TIME_process dont take the timerNext_us param hence cant return when is the next time to send the TIME trace.

Maybe its not necessary for the protocol to send exactly at the interval.

Is this a ok behaviour?

CANopenNode, 2023-11-21:
I think, exact intervals are not necessary. However time information is precise.
For synchronization there is a SYNC message.

CANopenNode, 2023-11-21:
But if necessary, timerNext_us could be added.

danirpg, 2023-11-22:
> I think, exact intervals are not necessary. However time information is precise. For synchronization there is a SYNC message.

Yeah the time in the data is precise so i guess its not critical.
Dont bother then.
Thanks

closed 2023-11-22 by danirpg