CANopenNode/CANopenNode #485: Dynamic Manufacturer-specific Object creation

mEft1998, 2023-12-10:
Hi, thanks for excellent & powerful CANOpen library & thanks for introducing the EDS Editor software.  Nowadays I use them to do my project but I have a problem: How to adjust the master object dictionary according to the behavior and settings of the slaves? Especially creation of a Manufacturer-specific (data) object in runtime (without using EDS Editor).
 Thanks a lot

CANopenNode, 2023-12-11:
Current approach is to provide enough enough empty OD variables, where data from remote nodes can be mapped. It should be possible to dynamically add variables to Object dictionary, but I don't know for examples.

Another option:
- create OD object(s) - arrays - at specific indexes with EDS editor, but don't set default values, so object(s) won't use any memory.
- Use OD extension and provide own read/write functions on OD access. You have full control, how variables are created.
- This is just an idea.

mEft1998, 2023-12-11:
I'll try to use your idea or provide enough empty OD variables. Thank you for your attention