CANopenNode/CANopenNode #489: On STM32,Multiple TPDOs uses event-driven transmission,  How to use the function "OD_requestTPDO(uint8_t *flagsPDO, uint8_t subIndex)" and "uint8_t *OD_getFlagsPDO(OD_entry_t *entry)" sends TPDO?

BZU2020, 2024-01-02:


mEft1998, 2024-01-06:
I think that you should use "CO_TPDOsendRequest(TPDO[idx])"

CANopenNode, 2024-01-08:
https://github.com/CANopenNode/CANopenDemo/blob/master/demo/CO_application.c#L112

CANopenNode, 2024-08-14:
> But I'm not really sure: if I have e.g. 16 temperature sensors which I have set that can be mapped to TPDO, do I have to follow this approach for every of the sensors? if this the intended method?

Yes. With analog values you could also use some hysteresis this way. But with event driven PDOs you may also set some long interval on event timer and also set inhibit timer.

Event driven PDOs may be very fast in CANopen, but they could also overload the traffic. So more reliable and deterministic approach could be with SYNChronous PDOs.

C0ntroller, 2024-12-06:
> Event driven PDOs may be very fast in CANopen, but they could also overload the traffic.

But isn't this exactly the point of the inhibit time? 
Also if we are talking about bus load, this is heavily dependent on your use case. A temperature will not change often, and sync telegrams and responses of the same value over and over, is way more load for the bus than.

---
So the current process seems less than optimal: 
1. I have to add an "extension" to all my mapped OD entries. To add some flags. But just knowing all mapped entries can be hard, as this even could change at runtime.
2. I have to manually keep track of changes on the mapped objects.

So, the only thing TPDOs currently do well, are timed and synchronous sending. Which is not working in event driven environment.

Why not just have another (`uint64_t`, as this is at max 8 Byte long) field in every TPDO object, that stores the last transmitted value that get's compared to the current state? So basically what every user now has to do for every mapped value individually?

CANopenNode, 2024-12-08:
> > Event driven PDOs may be very fast in CANopen, but they could also overload the traffic.
> 
> But isn't this exactly the point of the inhibit time? Also if we are talking about bus load, this is heavily dependent on your use case.

Yes, everything depends on use case. But inhibit time may not prevent overload. Synchronous transmission is more deterministic. For example, if some exceptional event happens on whole system, devices may try to send many different PDOs at the same time. And this may overload the bus at the very inconvenient moment. In case of synchronous PDOs there is no problem. However, that is just one point of view.

> So the current process seems less than optimal... Why not just have another (uint64_t, as this is at max 8 Byte long) field in every TPDO object, that stores the last transmitted value that get's compared to the current state?

That may be true for one user case, when we want to automatically send all or some PDOs after they change. That is quite simple solution, which was implemented in v1.3. But application could have no control over it.

"New approach" is current, where application has full control, not over the TPDO, but over the OD variable, which may be mappded to a TPDO...

C0ntroller, 2024-12-08:
> For example, if some exceptional event happens on whole system, devices may try to send many different PDOs at the same time. And this may overload the bus at the very inconvenient moment. In case of synchronous PDOs there is no problem. 

Yes, both have up- and downsides, depending on the use case. My prior comment was about overload from one single client, but you are totally right.

> That may be true for one user case, when we want to automatically send all or some PDOs after they change.

Imo this is not a "use case" but indicated by the "event driven" transmission types of the TPDO (254 and 255). 
 
> "New approach" is current, where application has full control, not over the TPDO, but over the OD variable, which may be mappded to a TPDO...

Well of course it's not working now, but I don't see how this makes implementing this harder or impossible? In the TPDO_process routine, just check the last sent value against the current value (calculated from the mapped OD entries). This would remove the need for flags on every mapped value and the need to mark OD entries changed in the first place.

So just some pseudo code. Just to get my idea across:
```c
TPDO_process() {
	for (TPDO in all TPDOs) {
		//.... checking sync flags, count down timers etc.
		uint64_t currentValue =  getCurrentTPDOValue(TPDO);
		if (TPDO.lastSentValue != currentValue && inhibitTime == 0) {
			sendTPDO(TPDO);
		}
	}
}

getCurrentTPDOValue(TPDO) {
	// Probably I switched LSB and MSB here or did some weird inbetween
	uint64_t current = 0;
	uint8_t index = 8;
	for (mappedEntry in TPDO) {
		index = index - mappedEntry.size;
		current |= mappedEntry.value << index;
	}
}
```

Can you tell me, why this couldn't work?

glmnet, 2024-12-09:
For sure your proposal could work, actually he just tell you it was how it worked before.

I'm pretty sure that a conditional compile flag and/or additional TPDO internal config could be set for this behavior to be implemented in the library and hence no TPDO request flag should be set. May be both methods can co-exist

This should have a table of pros and cons of each memory, mostly:
* Control over OD and TPDO: I still do not understand clearly what @CANopenNode means with this but seems most important
* OD Read function (probably related to previous point)
* Memory (the extension memory would be saved maybe but this certainly will add some other memory? plus the 64 bits to store last sent value)
* Performance (building the value to compare on every TPDO process will add a bit of overhead)

C0ntroller, 2024-12-11:
> What is actual problem with the current implementation? 

Everything about event driven TPDOs:
1. @glmnet hypothetical szenario: When having a lot of TPDOs, and values, storing the data and comparing the change is a lot of logic, but pretty redundant.
2. Event driven TPDOs currently are pretty static. You can't just put any OD index into the mapped values, because the OD entry needs to have the extension preconfigured. And value changes must be observed by the application, so the application has to know to mapped values and also update the observer when the mapping changes at runtime. Configuring event driven TPDOs dynamically is absolutly not trivial.

> Events are not "automatic", they requires some programming effort. Is this the problem? 

If you want to call it that, yes. Because timers are also not "automatic", yet this library provides functionality for them. 

> We could add also the alternate method. But, will the alternate method spend less processing power? ("getCurrentTPDOValue" is more complex as in example shown above.)

Probably more processing necessary but less development time and less CANOpen function specific code inside my application (what is, imo, the main reason to use a library like yours - not having to care about function or technology specifics).

I guess another, a bit less processing intesive way, would be to provide an OD entry write function, that checks for difference and set's a flag. As I understand this, this is already possible using these extensions, but imo that's not really intuitive (and the documentation about extensions even warns you about their powerfulness).

C0ntroller, 2024-12-18:
I finally got arround to create a custom write function which does the check automatically. Maybe this also will help someone in the future.  
But @CANopenNode could you take a look at it? It feels kind of verbose and assumes, you add the OD_entry_t* as object in the OD_extension_t for the entry. Is there a better way to request the TPDO in a custom write function without being vulnurable for future changes in the library?

```c
ODR_t OD_writeTPDOCheckChange(OD_stream_t* stream, const void* buf, OD_size_t count, OD_size_t* countWritten) {
    // Only stream is currently relevant for us. All other values will be checked in original method
    if (stream == NULL) return ODR_DEV_INCOMPAT;

    // Check if original data exists
    if (stream->dataOrig == NULL) return ODR_SUB_NOT_EXIST;

    // Copy original data in temporary buffer
    uint8_t* dataOrig = malloc(sizeof(uint8_t) * stream->dataLength);
    memcpy(dataOrig, stream->dataOrig, stream->dataLength);

    // Call original write function
    ODR_t ret = OD_writeOriginal(stream, buf, count, countWritten);

    // If it wasn't successfull we don't need to bother
    if (ret != ODR_OK) {
        // Before returning, we need to free the temporary buffer
        free(dataOrig);
        return ret;
    }

    // Get or find the OD_entry_t*
    OD_entry_t* entry = stream->object;
    if (entry == NULL) {
        entry = OD_find(OD, stream->index);
        // Still can't find it? Leave.
        if (entry == NULL) {
            free(dataOrig);
            return ODR_OK;
        }
    }

    // Check if the value changed and request TPDO if it is
    if (memcmp(dataOrig, buf, stream->dataLength) != 0) {
        OD_requestTPDO(entry, stream->subIndex);
    }

    free(dataOrig);
    return ODR_OK;
}
```

robincornelius, 2024-12-19:
Hi,

Can i ask why you are doing malloc(), memcpy() then memcmp(), is there a
reason you cannot simply memcpy(stream->dataOrig,buf,stream->dataLength)?
Would then reduce the malloc/free calls as these will be called a lot. I
may have missed something but a quick glance looks like there is an
optimisation.

CANopenNode, 2025-01-03:
If there is a need that CANopenNode detects detect change of state of the whole TPDO and sends it automatically, I would add this functionality directly into the TPDO object. Without extension and application interaction. This functionality could coexist and could be optionally enabled by configuration.

[CO_TPDOsend](https://github.com/CANopenNode/CANopenNode/blob/6ef8956ff92b28c2ede330f6f60c37b4b43c58e2/301/CO_PDO.c#L1127) function could be changed so, it would internally check for change of state for the whole TPDO. It could accept additional argument and be called from CO_TPDO_process() appropriatelly.