CANopenNode/CANopenNode #492: Send to different Nodes with same objects

TimK456, 2024-01-23:
Hi Guys,

I'm using the library to configure a canOpen node as master. It should communicate with to identical slaves (NodeID: 0x01 and 0x02). 
I used the canOpenEditor and the correspond EDS file of the slave devices to configurate the OD so far. I send the messages by using the event timer e.g. every 50ms and update and read the DOs by using the following both functions:

	ODR_t OD_get_u64(const OD_entry_t *entry, uint8_t subIndex, uint64_t *val, bool_t odOrig)

	ODR_t OD_set_u8(const OD_entry_t *entry, uint8_t subIndex, uint8_t val, bool_t odOrig)

That works fine with only one slave device. So if I have understood correctly, the reading of the messages depends on the CAN ID, which is made up of COB-ID + NodeID. However, if I only ever update the content of the objects here, I cannot address the different nodes differently. So there should be a function / possibility to address the objects and the corresponding nodes.

I'm a bit stuck on how to implement this. It would be nice if someone here could help me. I'm using a STM and also the CanOpenSTM32 library.

Thank you.

martinwag, 2024-01-23:
Object Entries are not directly linked to addresses. You need to use PDOs for this (TPDO for transmitter, RPDO for receiver(s) ). If you do this correctly, updating the value will lead to TPDO transmission, which in turn will update on all devices with configured RPDO.

TimK456, 2024-01-30:
Okey but I configure the Tx RX PDO like in the picture.

[image]

I think that's waht you mean, but if I update the object now, it will send to both devices and I will send different values to each device e.g. 0x01 to device 1 and 0x02 to device 2 on the same time.

martinwag, 2024-01-30:
you need to configure txpdo in sending device (0x1800 registers) and rxpdo in receiving device (0x1600 registers). for two independent transmission you need two tpdo in sending device (e.g. 181+1800 and 182+1801) and one rpdo each in receiving device (e.g. rx1: 181 + 1600, rx2: 182 + 1600)

BenjaminLangeEmbedded, 2024-01-31:
I probably have a similar problem as Tim if I understand correctly or at least the part he will come to next.
I have 4 motors of the same kind and i want to controll them independently. They all use the same mapping and I have no way to change it.

I can controll each motor alone when setting the eds file and change the values of the object. I also understand that i need to make 4 txpdo's with the COB-ID and Index (e.g. 181+1800, 182+1801, 183+1802, 184+1803). But how can I differentiate the objects? All 4 of them work with 0x6040 for the controlword. Can you point out how to change the value independently or isn't this possible and I have to change the value send txpdo1, change value send txpdo2 and so on?

Really appreciate your help.

martinwag, 2024-01-31:
@BenjaminLangeEmbedded sprichtst du deutsch, das bekomme ich auf enlisch nicht erklärt... 
- Mit dem R/TPDO Mapping erstellst du eine Verknüpfungstabelle zwischen Objekteinträgen in zwei Geräten. Eines davon (Motoren) kann eine feste Konfiguration haben. Das kannst du auf der anderen Seite durch entsprechendes Mapping geraderichten (ist CANopen Seitig auch so gedacht).
- Du kannst dein Mapping jetzt so einstellen, dass ein beliebiger Objekteintrag von deinem Gerät auf den fest vorgegebenen Eintrag auf dem anderen Gerät übertragen wird.
Bsp: Deine 4 Motoren möchten jeweils 0x6040 auf dem ersten Byte des PDO sehen. Du erstellst nun 4 TPDOs, der erste mappt in deinem lokalen OD 0x2000, der zweite 0x2001, usw. jeweils auf das erste Byte des TPDO. Jeder der TPDOs bekommt eine COB-ID, diese ist im lokalen Gerät und in den Motoren einzustellen. Das korrekte RPDO Mapping im Motor ist bereits fest vorgegeben. Der PDO Mechanismus sorgt nun automatisch dafür, dass jede Änderung in 0x2000 im Motor 1 in 0x6040, 0x2001 in Motor 2 in 6040 usw übernommen wird.

english machine translation:
With R/TPDO mapping, you create a linking table between object entries in two devices. One of them (motors) can have a fixed configuration. You can align that on the other side through corresponding mapping (similarly intended in CANopen).

Now, configure your mapping so that any object entry from your device is transmitted to the predefined entry on the other device.
Example: Your 4 motors each want to see 0x6040 at the first byte of the PDO. You now create 4 TPDOs, the first maps in your local OD 0x2000, the second 0x2001, etc., each to the first byte of the TPDO. Each of the TPDOs gets a COB-ID, which is to be set in the local device and in the motors. The correct RPDO mapping in the motor is already predefined. The PDO mechanism now automatically ensures that any change in 0x2000 in Motor 1 is transferred to 0x6040, 0x2001 in Motor 2 to 0x6040, and so on.

BenjaminLangeEmbedded, 2024-01-31:
@martinwag Das passt ja gut :) Erstmal vielen Dank für deine Hilfe.
CanOpen ist neu für mich und hier stoße ich auch vom programmieren her an meine Grenze daher echt nett von dir. 
Ich habe jetzt mal im Editor die Objekte erstellt. Node-ID dann dementsprechen 1,2,3,4. Passt das so?  

Und dann bin ich mir auch unsicher beim Zugriff auf die Parameter/Variablen. Wenn ich dich richtig verstehen liegen diese jetzt bei den TPDO's bei 0x1A00 bis 0x1A0F  und für die RPDO's bei 0x1600 bis 0x1607 richtig?

Wie greife ich dann auf diese zu? 
Geht das auch über OD_set oder OD_get also OD_set_u32(OD_find(OD, 0x1A00), 0x00, val, odOrig) ? Aber wie komme ich dann bei diesem an die 3 verschiedenen Einträge ran (0x6081, 0x6083, 0x6084)? 

Wäre super wenn du mir hier noch einen Tipp geben kannst. 

Translation:

 Firstly, thank you very much for your help.
CanOpen is new to me and I'm reaching my programming limits here, so that's really nice of you. 
I have now created the objects in the editor. Node ID then corresponds to 1,2,3,4. does that fit?  

And then I'm also unsure about accessing the parameters/variables. If I understand you correctly, these are now at 0x1A00 to 0x1A0F for the TPDOs and at 0x1600 to 0x1607 for the RPDOs, right?

How do I access them then? 
Is this also possible via OD_set or OD_get, i.e. OD_set_u32(OD_find(OD, 0x1A00), 0x00, val, odOrig) ? But how do I get to the 3 different entries (0x6081, 0x6083, 0x6084)? 

Would be great if you could give me a tip here. 

TX Motors (manufacturer):
[image]
RX Device (me) :
[image]

RX Motors (manufacturer):
[image]
TX Device (me):
[image]

BenjaminLangeEmbedded, 2024-01-31:
Or is it better to write directly? 
Like: 
OD_RAM.x1A00_TPDOMappingParameter.applicationObject1 = xxx;
OD_RAM.x1A00_TPDOMappingParameter.applicationObject2 = xxx;

Can't get my hands on the hardware until tomorrow to test it. 
Thanks again.

martinwag, 2024-01-31:
If you are completely new to CANopen, I would recommend having a look at "embedded networking with can and canopen" (https://www.canopenbook.com/) this explains SDO/PDO stuff quite extensively. I don't have time to go trough your post right now, sorry.

sicrisembay, 2024-02-05:
In DSP402 (v2 26-July-2002),  PDO's from different motor (axle) has offset of 64.  In this proposal, up to 8 motor (axle) controllers can be supported.

[image]

Note: I don't have access to the official CiA-402 specification document.  This is definitely outdated.

best regards,
Sicris