CANopenNode/CANopenNode #493: How to handle SIGSEGV during the communication process to prevent program crashes

Lemonononon, 2024-02-02:
Thank you very much for your outstanding work! 
I use PCAN to connect the computer (Ubuntu OS) to a motor. The recent issue I've encountered is that when the communication is unstable (here, unstable communication refers to the possibility of disconnection between PCAN and the motor, which is a problem I am investigating), CanOpenNode will directly throw a SIGSEGV ( Process finished with exit code 139 (interrupted by signal 11:SIGSEGV) ) and terminate the program. I have reproduced this issue by disconnecting PCAN from the motor during the program's execution. I am trying to figure out how to handle this error to prevent the program from crashing, but instead have the opportunity to handle and report this error.

martinwag, 2024-02-02:
You can't handle sigsegv, your program will always be terminated. You need to find and fix the place in the source code where the segmentation fault happens.

Lemonononon, 2024-02-03:
Thank you very much for your reply. In fact, my project is based on your CANopenLinux project. I have encapsulated it into a class and exposed the SDO and PDO related interfaces for upper-layer usage. I would greatly appreciate it if you could provide some debugging advice.

Lemonononon, 2024-02-03:
Oh, add one more point: I found that the error occurred here in clion debug mode.
The error reported in the log is: "(%s) Socket error msg ID: 0x%08x, Data[0..7]: 0x%02x, 0x%02x, 0x%02x, 0x%02x, 0x%02x, 0x%02x, 0x%02x, 0x%02x (%s)"
```
961 #if CO_DRIVER_ERROR_REPORTING > 0
962             CO_CANerror_rxMsgError(&interface->errorhandler, &msg);
963 #endif    
```
[CANopenLinux/blob/master/CO_driver.c](https://github.com/CANopenNode/CANopenLinux/blob/ebcc3c8f01c9360c3a090267611e805bc78bd236/CO_driver.c#L961-L964)

martinwag, 2024-02-05:
Run the stack in GDB (ide or plain), it will stop at the line that causes the segfault.