CANopenNode/CANopenNode #495: Multiple nodes, NMT problem

djsubzero, 2024-02-08:
Hi, 
I'm currently implementing a device (STM32) with multiple nodes (CO_MULTIPLE_OD).
I initialize the 3 nodes using co_config and each node having its own CO, NMT and OD instance.

The 3 nodes appear in the network (NMT Pre-op), but I can only start the last initialized node. 
After some digging in the code, I have found out that the object parameter in the CO_NMT_receive function is always the last one passed to CO_CANopenInit and thus not from the node to which the message was sent.

Am I missing something here? Or is this just not supported?

CANopenNode, 2024-02-09:
CANopenNode is implemented in object oriented way. So each element has its own data inside and any combinations should be possible.
However, files CANopen.h and CANopen.c links all objects and they are written for "simple" configuration. Those files are somehow complex because of flexibility.
For CO_MULTIPLE_OD I suggest to write own CANopen.h and CANopen.c. No need to complicate with all macros, but may write customized CO_CANopenInit(), etc. functions with adjusted set of arguments.

djsubzero, 2024-02-22:
Ok, thanks, I will try that