CANopenNode/CANopenNode #500: MISRA 2004 compliant for a few rules

NLI-BrianLinari, 2024-03-11:
Make MISRA compliant for
MISRA C 2004 rule 14.10
MISRA C 2004 rule 14.9
MISRA C 2004 rule 13.1
MISRA C 2004 rule 15.3

CANopenNode, 2024-03-11:
You should clone this repository, commit changes on your clone, then make a pull request here. Something like this.

NLI-BrianLinari, 2024-03-11:
I did clone it, commited localy then tried to push my branch but is said that i don't have access. I have to push the branch to make a pull request.
[image]

martinwag, 2024-03-12:
You should **create a fork on your own user account** clone this repository, commit changes on your clone, then make a pull request here. 
:-)
you also need to set-up your git client to work with github 2fa for this most likely...