CANopenNode/CANopenNode #502: Issue with include style fo Arduino

FStefanni, 2024-03-13:
Hi,

the current include style creates issues when trying to compile with Arduino.
For example, currently includes refers to a "base" directory:
```c
#include "301/CO_driver.h"
```

To simplify integration with Arduino, it should be relative, e.g.:
```c
#include "../301/CO_driver.h"
```

The only  exception is `CO_driver_target.h`, which can remain as is (`#include "CO_driver_target.h"`), or mutated to `#include <CO_driver_target.h>`, since it is not part of the library itself (i.e. it is implemented by the porting).

These changes are completely backward compatible.

Is it possible to perform these refinements?
In case, I am willing to propose a patch.

Thank you,
regards

martinwag, 2024-03-13:
as far as I know, using relative path for includes is considered "bad practice" in most cases. Normal use is to set the base dir as include dir in your makefile and then always to reference the base dir in your includes. This is also how the stack is used in the example: https://github.com/CANopenNode/CANopenNode/blob/master/example/Makefile

CANopenNode, 2024-03-13:
You should set include paths in Arduino somehow. Or manually copy files for your project.

FStefanni, 2024-03-13:
Hi,

it is not bad practice at all, since it has no downsides.

Arduino does no uses makefiles, it has a custom build system.
It is possible to pass somehow extra compiler flags, but it is not the usual way for Arduino (you just expect to install the library and everything working as is).

Regards