CANopenNode/CANopenNode #504: Trying to receive SDO messages in SDO client over the CO_SDOclient_receive

urrudelu30, 2024-03-14:
Hi, I am trying to receive messages over the RXFIFO in the SDO client, but every time a message is comming the function rejects its due to the SDO_C->state is in CO_SDO_ST_IDLE. How I can change this state, there is some function that I need to configure for starting receiving data?
Maybe I need to configure CO_SDOclientDownloadInitiate or CO_SDOclientUploadInitiate?

if (SDO_C->state != CO_SDO_ST_IDLE && DLC == 8U
&& (!CO_FLAG_READ(SDO_C->CANrxNew) || data[0] == 0x80)
) {

CANopenNode, 2024-03-14:
SDO client is complex, please read the docs or see CANopenLinux for implementation. Or try to use PDO instead SDO.

urrudelu30, 2024-03-14:
Hi, In this case I need to configure such a 50 SDOs in the CANopen bus between two devices. I am using an STM32 microcontroller with RTOS (no S.O Linux). Could you tell me, where the docs are?

CANopenNode, 2024-03-14:
CO_SDOclient.h contains quite useful information with example.
But make sure, you understand CANopen, SDO, PDO, etc. correctly: see the standard, wikipedia, internet, docs here, but the easiest way may be just reading canopenbook.com for example.

urrudelu30, 2024-03-21:
There is any manner to read a **int16_t** variable over **read_SDO()** function?, the function that returns the variable is a fifo that returns an array of uint8_t

int16_t overTemperatureLimit = 0;
read_SDO(SDO_C, NODE, 0x3020, 0x04, overTemperatureLimit, sizeof(overTemperatureLimit), &readSize);