CANopenNode/CANopenNode #507: Enter the pre-oprational state

Ashenoneq, 2024-03-22:
After startup, my device runs in the pre-operational state, and the NMT host controls me into the operational state. However, after running for a period of time, my device has entered the pre-operational state. I don't know why. What are the general conditions that will lead to pre-operational?
My NMT_CONTROL  Settings：
```
#define NMT_CONTROL  CO_ERR_REG_GENERIC_ERR \
                    | CO_ERR_REG_COMMUNICATION \
```

CANopenNode, 2024-03-22:
If error register is set, device enters pre-operational. See Emergency.

Ashenoneq, 2024-03-25:
> If error register is set, device enters pre-operational. See Emergency.

In the code, I only observed that the error register caused the system to enter the pre-operational state. However, when initializing CANopen and defining NMT_CONTROL, I did not enable CO_NMT_ERR_ON_BUSOFF_HB and CO_NMT_ERR_ON_ERR_REG.
```
/* verify NMT transitions based on error register */
    bool_t busOff_HB = (NMT->NMTcontrol & CO_NMT_ERR_ON_BUSOFF_HB) != 0
                    && (CO_isError(NMT->em, CO_EM_CAN_TX_BUS_OFF)
                        || CO_isError(NMT->em, CO_EM_HEARTBEAT_CONSUMER)
                        || CO_isError(NMT->em, CO_EM_HB_CONSUMER_REMOTE_RESET));
    bool_t errRegMasked = (NMT->NMTcontrol & CO_NMT_ERR_ON_ERR_REG) != 0
                    && (CO_getErrorRegister(NMT->em) & NMT->NMTcontrol) != 0;

    if (NMTstateCpy == CO_NMT_OPERATIONAL && (busOff_HB || errRegMasked)) {
        NMTstateCpy = (NMT->NMTcontrol & CO_NMT_ERR_TO_STOPPED) != 0
                    ? CO_NMT_STOPPED : CO_NMT_PRE_OPERATIONAL;
    }
    else if ((NMT->NMTcontrol & CO_NMT_ERR_FREE_TO_OPERATIONAL) != 0
        && NMTstateCpy == CO_NMT_PRE_OPERATIONAL && !busOff_HB && !errRegMasked
    ) {
        NMTstateCpy = CO_NMT_OPERATIONAL;
    }
```

CANopenNode, 2024-03-25:
There should be always emergency message when error register is set. See also object 0x1003 and candump. What is the reason for entering pre-operational?

Ashenoneq, 2024-03-28:
> There should be always emergency message when error register is set. See also object 0x1003 and candump. What is the reason for entering pre-operational?

The object 0x1003 is not enabled in my OD file.

closed 2024-06-07 by Ashenoneq