CANopenNode/CANopenNode #508: Please help

arthurodamasceno, 2024-03-29:
https://www.kalatec.com.br/wp-content/uploads/2021/10/INVT-SV-DA200-AC-Servo-Drive_CANopen-Technical-Guide.pdf

may i use this stack to control this servo driver? i'm totally lost, how do i setup TPDOS to send to servo driver commands?

arthurodamasceno, 2024-03-29:
Or should i use SDOs?

CANopenNode, 2024-03-29:
With SDO you configure parameters (drive parameters, configure PDOs, etc, store parameters). PDOs are used for exchange of data between CANopen devices. For configuration it is much easier, if you have a GUI tool. Usually manufacturers have something for their drives.

CANopenNode, 2024-03-29:
If you send correct PDOs (which are basically a raw CAN messages), you can run your drive.

closed 2024-04-08 by arthurodamasceno