CANopenNode/CANopenNode #519: SDO Client - Read request missing Index and subIndex byte.

RoopanJK, 2024-06-06:
Hi, I'm implementing the CANOpen stack in the new Arduino Rev 4 Minima board, using the reference of this [repository](https://github.com/thaanstad/ESP32_CanOpenNode/tree/main). I've managed to get the heartbeat and PDO messages working using OD. 

But when I tried to use this as a SDO client, the read request is sent but the index and subindex bytes are empty. So the SDO communication gets aborted. The data I'm trying to read is a 1 byte data.

I've used the function as below,
[image]

The sent CAN Messages
[image]

I've checked that the index is indeed updated in the SDO Client object within the `CO_SDOclientUploadInitiate` function. So what could be causing this?

CANopenNode, 2024-06-07:
SDO client is hard. I suggest using Arduino as simple CANopen device first. Try to use CANopenLinux (on raspberryPI or PC) as a client and for experimenting.

RoopanJK, 2024-06-10:
@CANopenNode Okay will do and get back to you if any updates.

RoopanJK, 2024-06-19:
The issue was on my driver file. I've forgot to assign rest of the bytes from the CanOpen stack buffer to my hardware message bytes.

closed 2024-06-19 by RoopanJK