CANopenNode/CANopenNode #523: Changing PDO mapping doesn't appear to follow ds301 remapping requirements.

ckhassan, 2024-06-12:
Hello,

According to ds301, RPDO mapping can be changed when the device is in NMT pre-operational or operational states.
In order to change the mapping, these steps are supposed to be followed:
1. Invalidate the RPDO: Set the RPDO index 1 invalid bit (i.e. set it to 1).
2. Disable mapping: Set the RPDO sub-index 0 to 0.
3. Modify the mapping as desired.
4. Enable mapping: Set the RPDO sub-index 0 to the number of mapped objects.
5. Validate the RPDO: Clear the RPDO index 1 invalid bit (i.e. set it to 0).
(Disclaimer: These five items are not quoted directly from ds301, but I did use ds301 as a reference.)

Does CANopen Node comply with ds301 on this (in which case, perhaps I'm just doing something incorrectly), or is it not implemented?

Many thanks!

CANopenNode, 2024-06-12:
Yes, this is called dynamic PDO mapping and is fully implemented as in CiA301.

ckhassan, 2024-06-12:
> Yes, this is called dynamic PDO mapping and is fully implemented as in CiA301.

Thank you very much for clarifying that, Janez.

closed 2024-06-12 by ckhassan