CANopenNode/CANopenNode #531: [feature] PDO bitwise mapping

Sl-Alex-KION, 2024-08-10:
Hello and thank you very much for this awesome project. I was using it in many projects, running on different microcontrollers and even on Android.
More than a half of the projects I'm working on use bitwise PDO mapping. I know that it is very easy to map the entire PDO to a single UINT64 object or to a union with a packed bit field, but this is not very convenient. Instead of having all the data defined in the object dictionary, we define them in the code.
I would like to volunteer and implement this feature. I see you invest o lot of time trying to reduce the amount of resources used by CANopenNode, so I'd do it in the following way:

1. Define ```CO_CONFIG_PDO_BITWISE_MAPPING``` in ```CO_config.h```, next to ```CO_CONFIG_PDO_OD_IO_ACCESS```. We still have some unused bits for this.
2. If this feature enabled, I'll reuse OD_IO.dataOffset to store the number of bits mapped instead of a number of bytes.
3. I'll take care about the following functions (hopefully didn't forget something, still need some time to debug all possible configurations):
    - ```PDOconfigMap```
    - ```PDO_initMapping```, both with and without ```CO_CONFIG_PDO_OD_IO_ACCESS```
    - ```OD_write_PDO_mapping```
    - ```CO_PDO_receive```
    - ```CO_RPDO_process```
    - ```CO_TPDO_process```
4. I'll take care of big-endian platforms. I don't have any big-endian HW to test my changes, but this should not be a big deal because PDOs are quite simple.

I already have some working skeleton I'd like to polish, test and share with you in a couple of weeks, so I have a couple of small questions:
- Would you accept such a PR?
- Do you plan any significant changes in the project (structure, code style, docs) in the nearest future?
- Is the idea with the additional config option fine for you, or you have another option?

Please let me know, it would be a real pleasure for me to make the perfect project even better. Even some commercial CANopen stacks don't have this feature.

Sl-Alex, 2024-08-10:
After this change is implemented, I'll need to add it to the Object Dictionary Editor, I already have a working skeleton there, so this could be a step 2.

CANopenNode, 2024-08-14:
> * Would you accept such a PR?

Yes, as long as PR is according to the standard, it is welcome.

> * Do you plan any significant changes in the project (structure, code style, docs) in the nearest future?

Recently there was significant changes (misra, reformatting) in the code, but there are no more similar plans. 

> * Is the idea with the additional config option fine for you, or you have another option?

Yes, additional config option is necessary and should be used as you describe. Code without `CO_CONFIG_PDO_BITWISE_MAPPING` will then be mostly unchanged.

< I see you invest o lot of time trying to reduce the amount of resources used by CANopenNode

Few years ago I had time to rewrite the stack. And so included many "config" options, so it could work also with small resources. And actually it runs on 16-bit microcontroller, 2kb RAM, 4R+4TPDO