CANopenNode/CANopenNode #532: Why dont I get all the Heartbeats of my sensors with a moderate busload?

NieLuc, 2024-08-19:
Hello :)
I am using a ESP32 with FreeRTOS and I have issues with heartbeats at a busload higher than 30 %. 
I know the heartbeats are sent at the correct time because I already checked that with Vector CANoe. Even with a HB Consumer Time of 250 ms and a producertime of 100 ms I get Heartbeaterrors.

CANopenNode, 2024-08-19:
ESP32 or FreeRTOS issues probably?

NieLuc, 2024-08-20:
The ESP32/FreeRTOS is often at IDLE so the computational power is enough. 
The `
        CANreceive(CO->CANmodule[0]);
function is called every millisecond. Could it be that messages get lost?

CANopenNode, 2024-08-20:
> Could it be that messages get lost?

Yes. But most likely not correct integration of CANopenNode into ESP32 and FreeRTOS.