CANopenNode/CANopenNode #534: Read and Write SDO index with error node

daddi1987, 2024-08-19:
Hello to all of you. Without putting the board containing the STM32 into Emergency, I am unable to read or write.
I request that the master node read from my node to the index 3001 00. The request is completed, and the node answers with the result, but it returns an error.
My slave node is 15(F)
[image]

daddi1987, 2024-08-20:
The Error 8120 Passive node Error caused by absent to other node in the network. but not undestended first error 00 00 00 01. 
do you have any ideas?

CANopenNode, 2024-08-21:
01 is a CO_EM_CAN_BUS_WARNING. Emergency is also sent on error reset with error code=0000.

Maybe just missing terminating resistor 120 ohm on the two ends?

daddi1987, 2024-08-21:
Yes the both termination Is on , dont have a problem using the PDO. Different with the SDO command evry command put the bord in panic mode.

CANopenNode, 2024-08-21:
SDO server is basic functionality and should work without special configuration. But it works in different thread than PDO...

If you started your application from scratch it may be complex to configure everything properly. I'm not familiar with STM32. Maybe you can first try with existing example, which should work. Then transfer it to your project. Or ask question there.