CANopenNode/CANopenNode #538: How to add PDO in non-consecutive Indexes

pirupic, 2024-09-04:
Hello 
I want to add a new PDO, 
but this PDO is not consecutive to indexes
0x1800, 0x1801, 0x1802, 0x1803,

is it possible to add, for instance a PDO in.... 0x190A index?

if not, what is the maximun number of PDO that I can use canopenNode library?

Thank you very much in advanced

CANopenNode, 2024-09-04:
Library itself is object oriented and it is possible almost anything.

The "problem" are files CANopen.c and .h, which tries to be flexible and so they may be complex. You can do some "magic" there. Or if you need customized solution, you can make your own.

You can activate CO_MULTIPLE_OD in your `CO_driver_target.h` or in Makefile. CANopenPIC (PIC32) or CANopenLinux or CANopenDemo contains most up to date examples (but not for CO_MULTIPLE_OD).

pirupic, 2024-09-05:
Yes, I think I could manage in CANOpen.c File. In deed, I did it in the previous CANOpen node version 6 years ago, but Now the actual code seems too different from the old, version, at some point it was a huge change.
For instance, Now I don´t know how to create CALLBACKs functions when I write in and index.
In the "old Style" Code there was a function called  CO_OD_configure that was easy to all and then implement the call back function
example: CO_OD_configure(CO->SDO, 0x2101       , CO_ODF_nodeId, 0, 0, 0);
and then 
you could create the call back function very easily code like this

static CO_SDO_abortCode_t CO_ODF_nodeId(CO_ODF_arg_t *ODF_arg){
}{

I cannot find how to make call backs in this code, neither in the PIC32 example, there is any call back.
do anyone  know where to find any example or documentation for the call backs?
thanks a lot
David.-

CANopenNode, 2024-09-05:
Yes, please see CANopenDemo project, there is example.

CANopenNode was actually rewritten. New approach to object dictionary as a core. All code was mostly rewritten. Configuration options allows low ram usage or comfort with functionalities. Documentation is up to date.