CANopenNode/CANopenNode #539: TPDO Error: Object Dictionary entry 0x1a0002

ruziev-dev, 2024-09-04:
Hi, I have faced with TPDO problem.

MCU: ESP32-C6

I`m trying to send TPDO like in a video example here (https://www.youtube.com/watch?v=R-r5qIOTjOo)

[image]
[image]

[image]
[image]

But there is some error in object Dictionary when `CO_CANopenInitPDO` is calling.

```c++
 err = CO_CANopenInitPDO(CO, CO->em, OD, activeNodeId, &errInfo);
        if (err != CO_ERROR_NO)
        {
            if (err == CO_ERROR_OD_PARAMETERS)
            {
                ESP_LOGE(TAG, "[CO_CANopenInitPDO]: Object Dictionary entry 0x%lx", errInfo);
            }
            else
            {
                ESP_LOGE(TAG, "PDO initialization failed: %d", err);
            }
        }

```

```log
I (731) CO_ESP32: Initializing
I (735) CO_ESP32: main task running.
I (739) CO_ESP32: Allocated 6640 bytes for CANopen objects
I (745) CO_ESP32: CANopenNode - Reset communication
I (751) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (760) gpio: GPIO[11]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (770) CO_driver: Driver installed
I (774) CO_driver: Driver started
I (778) CO_driver: Creating Tx Task
I (782) CO_driver: tx task running
I (786) CO_driver: Creating Rx Task
I (790) CO_driver: rx task running
E (795) CO_ESP32: [CO_CANopenInitPDO]: Object Dictionary entry 0x1a0002
I (801) CO_ESP32: creating periodic task
I (806) CO_ESP32: Periodic task running
I (811) CO_ESP32: periodic task created
I (815) CO_ESP32: CANopenNode is running
```

May be somebody could help me 
It's an .xpd file but for an opportunity to upload it to github I added ".txt" in end
[E_test.xpd.txt](https://github.com/user-attachments/files/16870140/E_test.xpd.txt)

CANopenNode, 2024-09-04:
Just clear "Count Label" parameter from all manufacturer and device profile parameters. 

Count Label "TPDO" is allowed only on objects 0x1800-0x19FF.

For more information see https://github.com/CANopenNode/CANopenNode/blob/master/doc/objectDictionary.md#object-dictionary-requirements-by-canopennode

CANopenNode, 2024-09-06:
Your device is pre-operational (0x7F). To see PDOs, it must be operational (0x05).

Se also https://github.com/CANopenNode/CANopenDemo

ruziev-dev, 2024-09-09:
@CANopenNode  Got it!

I tried to find reason why my node can't work on operational mode.

`x1001_errorRegister` shows me `16` as I understand it means communication error
```c++
 CO_ERR_REG_COMMUNICATION = 0x10U, /**< bit 4, communication error */
```

For reading  0x1003 I have written code
```c++
if (OD_RAM.x1001_errorRegister > 0)
            {
                OD_entry_t *errors_entry = OD_find(OD, 0x1003);

                uint8_t numbers_of_errors = 0;
                OD_get_u8(errors_entry, 0x00, &numbers_of_errors, false);

                if (numbers_of_errors > 0)
                {

                    printf("0x1003 ERROR DUMP:\n");

                    for (uint8_t i = 1; i <= numbers_of_errors; i++)
                    {
                        uint32_t error_dump = 0;
                        OD_get_u32(errors_entry, i, &error_dump, false);

                        printf("Error #%d: code: 0x%lx\n", i, error_dump);
                    }
                }
            }
```

It shows me :

```
0x1003 ERROR DUMP:
Error #1: code: 0x1a108200
```

bit 0-15 (0x8200) shows:
```c++
  /** 0x82xx, Protocol Error */
    CO_EMC_PROTOCOL_ERROR = 0x8200U,
```
What means bit 16-31 (0x1a10)?
How to interpretate it? 
And hoe to fix this problem?

ruziev-dev, 2024-09-11:
Ohhh... I fixed the error above but I have faced with new trouble. 

TPDO messages is constantly zero.
I'm changing variable from OD.

```c
 OD_RAM.x6000_test_1 = OD_RAM.x6000_test_1 + 5;
 printf("OD_RAM.x6000_test_1: 0x%lX\n", OD_RAM.x6000_test_1);
```

```
OD_RAM.x6000_test_1: 0x6
OD_RAM.x6000_test_1: 0xB
OD_RAM.x6000_test_1: 0x10
OD_RAM.x6000_test_1: 0x15
OD_RAM.x6000_test_1: 0x1A
```

This variable is mapped to TPDO

[image]

But PDO is constantly zero

[image]

How to solve it?